gazebosim/sdformat #253: proposal: Should determine how to specify a default configuration for joints

EricCousineau-TRI, 2020-04-23:
This was discussed between @joemasterjohn, @azeey, @scpeters, @sammy-tri, and myself earlier today (2020/04/23) in a VC. Transcribing that to an issue.

`//joint/axis/initial_position`  will be deprecated when SDFormat 1.8 comes out, because it's true purpose was not well-defined, appeared redundant w.r.t. `//world/state` and (primarily) was not used at all in Gazebo (or maybe anything else?):
https://github.com/osrf/sdformat/blame/f5c88a94/Changelog.md#L13

However, as discussed here:
https://github.com/RobotLocomotion/drake/issues/13065#issuecomment-617308431
There should be some way to specify a default configuration that is intrinsic to a model (especially for closed-loop configurations), but this *will not* affect the zero configuration, and will only be used in downstream applications. This ensures the definition is decoupled from context (encapsulation) and decouples  physical validity from modeling.

It is proposed to offer some way to specify this default configuration (ideally in minimal coordinates), for multiple types of joints (single-DoF, mutli-DoF), in a clear and concise syntax, and hopefully is termed in such a way that its purpose is *clearly* not intended to be something like `//world/state`.

FTR, currently, Gazebo closed-loop chain models require assembly (e.g. must be physically valid).

Some add'l things that the proposal should discuss:
* What `libsdformat`s responsibility is (i.e. parsing the value, storing it in DOM, and making it accessible later, suggested API)
* What downstream applications are responsible for to adhere to the spec (i.e. use it when a user asks for a *default configuration*, not to be confused with a  zero configuration()
* What should the syntax look like for all types of joints supported in SDFormat.

In talking with Joe, he said he would be up for taking a stab at submitting a PR to `sdf_tutorials` with the proposal.

Joe, if you would like help on something (or feel that you will not have time to do this), please let me know.
Here's the format for making a proposal:
http://sdformat.org/tutorials?tut=proposal_format

A PR should be made against https://github.com/osrf/sdf_tutorials

\cc @sherm1

EricCousineau-TRI, 2020-04-23:
Realized that the `sdf_tutorials` doc doesn't have anything on how to submit PRs. Made a stab at it here:
https://github.com/osrf/sdf_tutorials/pull/9

traversaro, 2020-04-23:
Beside the meaningful distinction between default and initial configuration, I think that the following (Classic) Gazebo issues are related: 
* https://github.com/osrf/gazebo/issues/1072
* https://github.com/osrf/gazebo/issues/230

sherm1, 2020-04-23:
BTW when I was working with John @hsu and @scpeters a few years ago we made some progress in generalizing sdf to support joints in their general, unassembled form which looks something like this:
[image]
(borrowed from Simbody; read "joint" rather than "mobilizer" here)
The joint definition includes a frame on the parent (F in the picture) and one on the child (M) and the meaning of q=0 is that F and M are coincident (typically). I'm not sure that work every made it into an official sdf spec though.

In general assembly must be seen as a computation (solve for a configuration that satisfies all constraints). It is actually a very difficult nonlinear problem, especially for 3d mechanisms with closed topologies. In the case of a design study that can change dimensions or placements of joints, assembly must be recomputed at each design step.

CAD systems do this in a very general way, allowing users to build "parts" and specify "mates" (and other constraints) and then assemble.

(I'm mentioning this so we don't accidentally lock in the special case where joints can always be defined in an assembled configuration in the sdf.)

EricCousineau-TRI, 2020-04-23:
> (I'm mentioning this so we don't accidentally lock in the special case where joints can always be defined in an assembled configuration in the sdf.)

Er, just to check, did you mean "[...] where joints can **only** be defined in an [...]"?
(Don't think that changes the meaning at all, but just wanna make sure!)

EricCousineau-TRI, 2020-04-23:
@traversaro osrf/gazebo#230 seems like it could be covered by `//world/state`? (will post in that issue)

sherm1, 2020-04-23:
> did you mean "[...] where joints can only be defined in an [...]"?

Yup, thanks -- edited.

traversaro, 2020-04-24:
> 
> 
> @traversaro [osrf/gazebo#230](https://github.com/osrf/gazebo/issues/230) seems like it could be covered by `//world/state`? (will post in that issue)

Probably yes, but my main goal was just to provide a backlink to this issue if someone found the old issue on Google.

EricCousineau-TRI, 2020-08-06:
Relates #138