gazebosim/sdformat #260: Make parsing of bad values (syntactically and semantically) more strict

azeey, 2020-04-29:
Example of bad syntax: #228 
Example of bad semantics: https://github.com/robotlocomotion/drake/pull/12863#pullrequestreview-375615059

This is actually a spec change; "bad" values should throw an error by design, not be ignored by design. Currently, a negative effort limit implies ignoring a joint limit...

For handling bad semantics, look into using `min` and `max` values in the spec descriptions

For non-scalar values like `//joint/axis/xyz`, libsdformat should enforce that the axis is a unit vector by either re-normalizing or requiring it to be normalized.

Also, spec should say that it will disregard non-unit vectors for axes.

We should do PR for libsdformat10 as an error.

If we have extra time, backport to libsdformat9, but as a warning. (With updates to Gazebo Classic + Ignition Gazebo.)

azeey, 2020-04-29:
Addressed by #244 and #259

EricCousineau-TRI, 2020-06-25:
Marking as done, but spun off separate issue about `//joint/axis/xyz`
https://github.com/osrf/sdformat/issues/307

closed 2020-06-25 by EricCousineau-TRI