gazebosim/sdformat #261: Supporting parsing "inf", "+inf", "-inf"

azeey, 2020-04-29:
To best enable: #249

scpeters, 2020-05-13:
I've added tests in #277 that confirm that `+Inf` and `-Inf` can already be parsed successfully.

scpeters, 2021-06-17:
I've just discovered that this doesn't work for elements that aren't part of the schema, such as elements within a `<plugin/>` block. The `typeName` for Element values that aren't part of the schema [defaults to `"string"`](https://github.com/osrf/sdformat/blob/sdformat11_11.1.0/src/parser.cc#L1489), and I've repeated the tests from #277 for elements with `"string"` `typeName` in https://github.com/osrf/sdformat/commit/9c7f328639e96b0d39e7a703f67d72096a29e774 (branch [scpeters:get_inf_from_string](https://github.com/osrf/sdformat/compare/sdf9...scpeters:get_inf_from_string?expand=1), and they fail.

scpeters, 2021-06-17:
I can create a separate ticket if you think I should leave this closed.

scpeters, 2021-06-17:
> I've just discovered that this doesn't work for elements that aren't part of the schema, such as elements within a `<plugin/>` block. The `typeName` for Element values that aren't part of the schema [defaults to `"string"`](https://github.com/osrf/sdformat/blob/sdformat11_11.1.0/src/parser.cc#L1489), and I've repeated the tests from #277 for elements with `"string"` `typeName` in [9c7f328](https://github.com/osrf/sdformat/commit/9c7f328639e96b0d39e7a703f67d72096a29e774) (branch [scpeters:get_inf_from_string](https://github.com/osrf/sdformat/compare/sdf9...scpeters:get_inf_from_string?expand=1), and they fail.

we discovered this during review of https://github.com/ignitionrobotics/ign-gazebo/pull/833#issuecomment-857069329

azeey, 2021-06-25:
@scpeters yes, I think a separate ticket would be better. I'll go ahead and create one.

azeey, 2021-06-25:
https://github.com/osrf/sdformat/issues/603

PR #277 (2020-05-28): Param_TEST: test parsing +Inf and -Inf
This adds a test confirming that "Inf", "+Inf', and "-Inf" and several lower-case and upper-case variations will parse to the `double` value of `std::numeric_limits<double>::infinity` or `-std::numeric_limits<double>::infinity` (as long as `std::numeric_limits<double>::is_iec559` is true).

Closes #261.
closed 2021-06-25 by azeey