gazebosim/sdformat #262: URDF -> SDFormat: Convert empty links to frames.

azeey, 2020-04-29:
It is common to create empty links to represent frames in URDF. However, when converting URDF to SDFormat, such links are ignored because they have zero mass. While this is fine for simulation, the frame information represented by empty links is lost. 

Now that frames are supported in SDFormat, we should create a frame for empty links and preserve their frame information for downstream users.

Relevant comments from Asana:
- @scpeters : links without inertias connected to another link via a fixed joint could clearly be converted to a frame on the parent link, but what if the link without inertia has a collision or visual, or it's connected by a different type of joint?

- @EricCousineau-TRI 
Sounds like using the welding approach work best, though it does incur the cost of an additional joint.
For visuals / collisions in those links, I guess you could update their //pose/@relative_to, but that sounds a tad hairy.

- @azeey 
So, currently, by default (you can set a parameter to disable this), the collisions and visuals get lumped into the link that has inertia. We can keep doing that, but in addition, create a frame for the the link without inertia attached to the link with the inertia. This would make it possible for a gazeb-ros plugin to publish static transforms for the link without inertia, which I think is the desired effect.

traversaro, 2020-04-29:
Related issue: https://github.com/osrf/sdformat/issues/199 .

EricCousineau-TRI, 2020-04-29:
Thanks!
This actually sounds like it's a duplicate of #199. If y'all think so, I vote for closing the older one?

EricCousineau-TRI, 2020-04-29:
Also, meta: Is it possible to word issues in such a way that they're not ambiguous with PR titles?
e.g. #199's title seems good; this one could be "issue"-ified by saying "Should convert [...]" rather than "Convert"?

scpeters, 2020-04-30:
there's a lot of discussion in #199; I'd prefer to close this and add these comments there

EricCousineau-TRI, 2020-04-30:
SGTM.

azeey, 2020-04-30:
Closed in favor of #199

closed 2020-04-30 by azeey