gazebosim/sdformat #263: Incremental SDF loading

chapulina, 2020-04-30:
Not all elements in an SDF file necessarily need to be loaded at startup. For example, Ignition Gazebo uses [`<level>`](https://ignitionrobotics.org/api/gazebo/3.0/levels.html) tags to separate what parts of the world will be loaded according to where robots are.

It would be nice of the SDFormat library supported a way of incrementally loading an SDF file into C++ objects for such use cases.

EricCousineau-TRI, 2020-04-30:
To check, this would just be for `//world/level`, not `//model/level`?

chapulina, 2020-04-30:
I just used `<level>` as an example use case, note that it isn't even part of the spec (we've been [hackily using it inside `<plugin>`](https://github.com/ignitionrobotics/ign-gazebo/blob/master/examples/worlds/levels.sdf#L4286)).

I can't imagine a use case for parts of a model to be loaded incrementally, but perhaps there's some. Our immediate use case would be for parts of the world.