gazebosim/sdformat #265: Should support libsdformat in robot_state_publisher

azeey, 2020-04-30:
Should extend robot_state_publisher to use libsdformat, parse a subset of SDFormat (KDL-compatible).

Should target:
https://github.com/ros/robot_state_publisher

EricCousineau-TRI, 2020-04-30:
Created per convo with @iantheengineer.

Some additional things to consider:
- Could be supported by having `libsdformat` depend on KDL (or whatever intermediate representation `robot_state_publisher` uses), and emit a structure 
- I haven't looked yet at how `robot_state_publisher` handles multi-dof joint names. We should ensure this is handled here as well.

(Possibly relates #253 in how joint dofs are named.)

\cc @joemasterjohn

EricCousineau-TRI, 2020-05-07:
Per f2f with @scpeters, @azeey, @IanTheEngineer:

My suggestion is to make users can "hook" / override URIs. `package://` should *only* be handled by ROS, not by some weird shim in `libsdformat`. (If there is a shim, it should get overridden.)

EricCousineau-TRI, 2020-05-07:
FTR The "weird" shim is the current "addURIPath" API:
https://github.com/osrf/sdformat/blob/sdformat9_9.0.0/include/sdf/SDFImpl.hh#L67-L72

A better mechanism would be to purely something like `setFindCallback`, but strictly associated with a given URL prefix:
https://github.com/osrf/sdformat/blob/sdformat9_9.0.0/include/sdf/SDFImpl.hh#L74-L79

IanTheEngineer, 2020-05-08:
It seems this task can be accomplished by the steps I've outline in 
https://github.com/ros/urdf/issues/34

Namely, a `sdformat_parser` ROS package that populates the `urdf::Model` API through `urdf::Model::initString()`

EricCousineau-TRI, 2021-01-11:
@sloretz Would you say this task is now done, given your most recent ROS2 contributions?

EricCousineau-TRI, 2021-01-11:
Sweet! Marking this as resolved.

closed 2021-01-11 by EricCousineau-TRI