gazebosim/sdformat #267: Non-parsed elements becoming (partially) parsed hurts existing downstream users?

EricCousineau-TRI, 2020-05-01:
Example issue pointed out here, which broke our functionality (upgrade from 9.1.x to 9.2.x):
https://github.com/RobotLocomotion/drake/pull/13201#issuecomment-622419190
In Drake, we were manually parsing `//collision/surface/friction`.

However, due to:
https://github.com/osrf/sdformat/blame/c6dac91c7845b0c45611e4db338e240bd86049d8/Changelog.md#L45-L46
https://github.com/osrf/sdformat/compare/a3468560e199a2723bf6ef544d338af82eea6871..76d1e4dc3257c6a529fa77b741b7fcdb67e0b1b6

For whatever reason, this same code path no longer works.
I currently cannot say why this has happened, as the raw XML should still be there (at least per the spec)?

I believe this is a bug, and should be fixed.

\cc @azeey @scpeters

scpeters, 2020-05-01:
I commented on the drake issue https://github.com/RobotLocomotion/drake/pull/13201#issuecomment-622613872

scpeters, 2020-05-02:
I've submitted a reversal of the recent behavior change in #268, but I think the drake code is brittle and should be improved.

scpeters, 2020-05-12:
I think the behavior change was fixed by #268 and upstream drake code is being updated as well

closed 2020-05-12 by scpeters