gazebosim/sdformat #272: Should check if gazebo URDF inclusion via SDFormat 1.6 is libsdformat issue

EricCousineau-TRI, 2020-05-07:
`sdformat`-tracking version for Gazebo issue:
https://github.com/osrf/gazebo/issues/2728

One proposal is to check behavior on Drake as well. If it is, it's a `libsdformat` issue, or a SDFormat 1.6/1.7 spec issue. If it is not, it's a Gazebo problem.

traversaro, 2020-05-07:
I forgot to mention this in https://github.com/osrf/gazebo/issues/2728, but I tried to convert both models (SDF 1.6 and 1.7) via `gz sdf -p model.sdf`, and the output was exactly the same, so I decided that the problem was probably on the Gazebo  side.

EricCousineau-TRI, 2020-06-11:
FTR https://github.com/osrf/gazebo/pull/2734 is the PR that could fix this

azeey, 2020-07-22:
Closed by osrf/gazebo#2734

closed 2020-07-22 by azeey