gazebosim/sdformat #274: Ruby dependencies may be undesirable?

EricCousineau-TRI, 2020-05-09:
Filing this issue based on:
https://github.com/RobotLocomotion/drake/issues/13231
https://github.com/ros/urdfdom_headers/issues/59#issuecomment-626147511

I'm not proposing any action yet, but rather to gather use cases where Ruby is high-overhead.

EricCousineau-TRI, 2020-05-09:
From https://github.com/ros/urdfdom_headers/issues/59#issuecomment-626147511:
> [...] but unfortunately it depends on Ruby [...]

@traversaro Just to check, what would a more ideal dependency be? Pure C++, Python, a more lightweight Ruby implementation, ...?

From prior conversations, I believe most Ignition tools use Ruby (see linked convos here):
https://github.com/RobotLocomotion/drake/issues/13231#issuecomment-624160257

traversaro, 2020-05-09:
> > [...] but unfortunately it depends on Ruby [...]
> 
> @traversaro Just to check, what would a more ideal dependency be? Pure C++, Python, a more lightweight Ruby implementation, ...?

Both C++ or Python would be much more convenient from the point of view of macOS/Windows use. Having a dependency on Ruby is relatively fine  as long as you don't want to load in it plugin written in C++: at that point, you will have problems if  your plugin  has  been compiled with a toolchain different from the one with which  the Ruby interpreter has been compiled.  You can find a few related discussions in:
* https://github.com/ignitionrobotics/ign-gazebo/issues/25
* https://github.com/ignitionrobotics/ign-tools/issues/7
* https://github.com/ignitionrobotics/ign-gazebo/issues/44

In theory you could  have  the same  problem in  Python as well, but in practice (at least on Windows) the Python interpreter uses the same C++ ABI of Visual Studio, that is the compiler used 
by most projects, and even if you want to recompiled the python interpreter  for mingw that is relativly easy.

EricCousineau-TRI, 2020-05-09:
Gotcha, thanks for the explanation!
And just to make sure I understand this workflow:
>  If instead you want to load in a Ruby script a plugin written in C++ [...]

Is this most directly addressed by https://github.com/ignitionrobotics/ign-tools/issues/7, where `ign` is the Ruby script that you mention which loads subcommands as shared library functions (rather than doing separate binaries like `git`, `rosrun`, etc. do for some of their more complex subcommands)?

traversaro, 2020-05-09:
> > If instead you want to load in a Ruby script a plugin written in C++ [...]
> 
> Is this most directly addressed by [ignitionrobotics/ign-tools#7](https://github.com/ignitionrobotics/ign-tools/issues/7), where `ign` is the Ruby script that you mention which loads subcommands as shared library functions (rather than doing separate binaries like `git`, `rosrun`, etc. do for some of their more complex subcommands)?

Exactly.

traversaro, 2020-10-30:
Another example in which the Ruby (build) dependency is cumbersome: https://github.com/conda-forge/libsdformat-feedstock/issues/9 . TL;DR: conda-forge removed ruby from their CI machine, and now the only "clean" way forward is to add a ruby port on conda-forge that builds on Windows.