gazebosim/sdformat #278: proposal: Better composition behavior?

EricCousineau-TRI, 2020-05-13:
This is a tracking issue for Model Composition.

Currently at:
http://sdformat.org/tutorials?tut=composition_proposal ([permalink](https://github.com/osrf/sdf_tutorials/blob/cf72a5aad51f920777b501df11ceed31928a68fe/composition/proposal.md))

chapulina, 2020-06-07:
I noticed that the current proposal is focused on poses. So this may be out of scope for this task, but I'd like to mention that an important aspect of composition is injecting plugins into SDF files.

The proposal says that `//include/plugin` is generally Gazebo-specific, but this is a concept we've also adopted into Ignition. My immediate use case is to have a model described once, and augment it with simulator-specific plugins as needed. Using the `<include>` tag, this is already possible for:

* `//model/plugin`
* `//actor/plugin`

But not for:

* `//model/link/sensor/plugin`
* `//model/link/visual/plugin`

EricCousineau-TRI, 2020-06-11:
> Using the `<include>` tag, this is already possible for [...]

Can you file a PR against the legacy Composition tutorial to make this existing contract explicit? (e.g. point out where in Gazebo it's supported, and other downstream *applications* if applicable?)
http://sdformat.org/tutorials?tut=composition&cat=specification&

---

At a (potentially distracting) "philosophical" level:

It's unclear to me what "adopted into Ignition" *really* means - can you point to other use cases of  `//plugin` outside of Gazebo / its upstream libraries?

FTR, my understanding of Ignition is that it aims to be individual libraries, but to me it still *smells* like it's application layers of Gazebo just broken out into separate repos. (At least, that is how it has been working with SDFormat, in both the software library and the specification.)

Totes down for it evolving it to that point, but I just want to make sure that things aren't claimed to be individual libraries if they aren't yet, and that we take the explicit steps to take them away from application layers of Gazebo to actual libraries?

EricCousineau-TRI, 2020-06-11:
Uh oh. I must eat my own words.

I am completely incorrect; it's in the documentation here as @scpeters pointed out:
http://sdformat.org/tutorials?tut=composition&cat=specification&#including-a-model

That being said, I would still love to have this feature be tracked.
Talking with @scpeters, he will open up an issue to track this.

Sorry for the noise!!!

EricCousineau-TRI, 2020-06-11:
Talked it more over w/ @scpeters and @azeey, and @scpeters will file an issue about injecting `//plugin` (and even custom tags) on elements; still dunno if it should be physical or just XML coordinates.

scpeters, 2020-06-17:
updating proposal with details on parsing implementation in https://github.com/osrf/sdf_tutorials/pull/28

EricCousineau-TRI, 2020-10-01:
Posted announcements:
* https://discourse.ros.org/t/sdformat-1-8-composition-proposal/16630/2
* https://community.gazebosim.org/t/sdformat-1-8-composition-proposal-cross-post-with-ros-discourse/667

chapulina, 2020-10-07:
I just noticed the follow up questions now and I'm not sure about where the conversation currently stands, so I'll just clarify some things.

> It's unclear to me what "adopted into Ignition" really means - can you point to other use cases of //plugin outside of Gazebo / its upstream libraries?

This issue has some examples: https://github.com/osrf/sdformat/issues/93

> my understanding of Ignition is that it aims to be individual libraries, but to me it still smells like it's application/framework layers of Gazebo just broken out into separate repos

The separate libraries are useful on their own without Gazebo. The same way you're using SDF outside Gazebo, there are projects using Ignition Rendering, GUI, Sensors, Math, Common, Math, Transport... Some public examples are [tesseract-ignition](https://snapcraft.io/tesseract-ignition) and [ign-rviz](https://github.com/ignitionrobotics/ign-rviz/).

> explicit steps to take them away from application layers of Gazebo to actual libraries?

We're all ears. What explicit steps are needed for them to smell like actual libraries to you? Perhaps this is not the right place to follow up on this discussion, but you're welcome to start a conversation on https://community.gazebosim.org/.

chapulina, 2020-10-07:
And the reason I ended up in this issue again was to cross-reference this design proposal for Gazebo, which I think is relevant to this issue. It talks about composition beyond poses, with the addition, removal or modification of included models.

https://github.com/ignitionrobotics/design/pull/5

EricCousineau-TRI, 2020-10-08:
\cc @claireyywang

azeey, 2021-08-10:
The last PR on this epic has now been merged, so I think we can close this. @EricCousineau-TRI ?

EricCousineau-TRI, 2021-08-12:
Sounds great!!!

closed 2021-08-12 by EricCousineau-TRI