gazebosim/sdformat #282: Figure out how to build FrameAttachedToGraph for static models with no links

scpeters, 2020-05-15:
In the initial version of the frame semantics proposal, all models were required to have at least one link, so that they could have a canonical link to which the implicit model frame (`__model__`) would be attached. This requirement was relaxed in the proposal in [Bitbucket sdf_tutorials PR 45](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdf_tutorials/pull-requests/45/page/1) (https://github.com/osrf/sdf_tutorials/commit/dea8c403b9b896ead466aa890d0aacd9d37914d2) so that static models would not be required to have a link with the rationale that its frames could be attached directly to the world. This was manifested in libsdformat with [Bitbucket sdformat pull request 626](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/626/page/1) (https://github.com/osrf/sdformat/commit/4fa3f81e16c7c0312d629686b7cb1f0d209655f5?w=1), which stopped reporting errors for static models without a single link. Eliminating that error was the primary goal of that PR, and rather than also modify `buildFrameAttachedToGraph` to handle static models without links in that PR, it just skips building the `FrameAttachedToGraph` for all static models regardless of the number of links they have.

So, figure out how to build FrameAttachedToGraph for static models with no links.

scpeters, 2020-05-28:
See also some initial thoughts here: https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/626/page/1#comment-127736189

azeey, 2020-12-02:
Can this be closed since #381 builds the FrameAttachedToGraph for static models?

scpeters, 2020-12-02:
sounds reasonable to me

closed 2020-12-02 by scpeters