gazebosim/sdformat #283: Nested models should work via DOM in `libsdformat 10`

EricCousineau-TRI, 2020-05-15:
At present, models that are *directly* nested in a `*.sdf` file are unable to be parsed:
https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/610/page/1

They should be supported, e.g. via `Model::NestedModels()`, or something akin to that.

scpeters, 2020-05-21:
I've started work on this in the following branch:

* https://github.com/scpeters/sdformat/tree/nested_model_dom

scpeters, 2020-05-21:
this implies a change in the parsing stages, so I'd like to make update those in the documentation as well (see #280)

scpeters, 2020-07-01:
I've implemented the `Model::Model*` nested model accessors in a draft PR (https://github.com/scpeters/sdformat/pull/1). It loads nested model DOM objects but doesn't add support for frame semantics like the following:

~~~
<model name="outside_model">
  <link name="middle_link"/>
  <model name="middle_model">
    <link name="inner_link"/>
  </model>
  <frame name="middle_frame" attached_to="middle_model"/>
</model>    
~~~

scpeters, 2020-07-06:
I was planning to submit a PR about this after #304 lands, but after discussing it with @azeey, it seems like we could target this to `libsdformat9` / SDFormat 1.7. We will discuss with @EricCousineau-TRI

scpeters, 2020-07-17:
since `libsdformat9` has not been included in Ubuntu 20.04, I think we have a chance to add this behavior to libsdformat 9.3 in time for 20.10

EricCousineau-TRI, 2020-08-20:
Marking as closed. Farmed out canonical links to #343 and #342

closed 2020-08-20 by EricCousineau-TRI