gazebosim/sdformat #287: Permit detecting an element / attribute was explicitly set (user) or implicitly set (conversion / default value)

EricCousineau-TRI, 2020-05-16:
From discussion on 2020-01-23 on Asana.

Notes:
Default values
* Know if a property was actually defined
* `libsdformat`, in its conversion phase, substitutes in default values
* Odd / non-strict value parsing (e.g. colors, RGBA)
* API: Could add an attribute check, `IsExplicitlySet()` or `UsedDefaultValue()` or something, and have overloads for querying that uses `optional<>` or returns a flag
* May be hard to check if a default value was actually set?
* Could be useful for deprecating attributes that a user set?

\cc @SeanCurtis-TRI

EricCousineau-TRI, 2021-04-08:
\cc @marcoag

azeey, 2021-06-03:
Waiting for https://github.com/osrf/sdformat/pull/575#issuecomment-845191669 to be resolved before closing.

marcoag, 2021-06-28:
Resolved: https://github.com/osrf/sdformat/pull/575#issuecomment-869458363. Closing this issue then.

closed 2021-06-28 by marcoag