gazebosim/sdformat #293: Allow "drilling down" into nested model frames

EricCousineau-TRI, 2020-05-28:
Towards #278

Should happen after #283

Some options:
- Just "hoist" duplicates of the frames (and frame graphs) with `{model_name}::{frame_name}`
- Actually try to parse the name and actually navigate down

scpeters, 2020-08-11:
Some User Story ideas for this feature:

* Loading a model that contains `::` syntax for referencing frames inside a nested model in the following places:
    * `//joint/parent`
    * `//joint/child`
    * `//frame/@attached_to`
    * `//pose/@relative_to`
* Resolving a `SemanticPose` to a nested frame (i.e. `link->SemanticPose.Resolve(pose, "nestedModel::nestedLink");`)
* Resolving the attached-to body for a frame attached to a nested link (i.e. it returns `"nestedModel::nestedLink"`)
* Resolving the attached-to body for a frame attached to a nested model to the canonical link of that nested model
* Resolving a joint parent/child to a nested link (i.e. it returns `"nestedModel::nestedLink"`)
* Support passing scoped names to the following functions:
    * Model::ModelByName
    * Model::LinkByName
    * Model::JointByName
    * Model::FrameByName
* Getting a `SemanticPose` object corresponding to a scoped frame name (i.e. `Errors Model::SemanticPoseByName(std::unique_ptr<SemanticPose> &_semanticPose, const std::string _scopedFrameName)`). We could actually implement an unscoped version of this in `libsdformat9` that just checks the Links, Joints, and Frames in the current model for matches with the passed name and returns the SemanticPose of the named object.

scpeters, 2020-08-12:
> Support passing scoped names to the following functions:
> 
>     * Model::ModelByName
> 
>     * Model::LinkByName
> 
>     * Model::JointByName
> 
>     * Model::FrameByName

I've prototyped a variation on this in https://github.com/scpeters/sdformat/commit/9eec062d048abbe5f70b3d12c69d07b4cbd0f7eb by adding `Model::*ByNestedName` and `Model::*NestedNameExists` methods that accept a `::` delimited name while leaving the existing APIs unchanged.

EricCousineau-TRI, 2020-08-13:
For supergraph + subgraph assembly, scoping, and queries, options for recursion:
- Copy + Graft
- Scoped Alias + Propagate
Pros/cons are in maintaining invariants and ease of local checks (e.g. encapsulation at each level).

For element queries, e.g. `LinkByName` vs. `LinkByNestedName`, `LinkByName` being able to resolve with "::" may be the best for backwards compatibility (the legacy composition behavior for include) as well as being able to easily resolve elements (e.g. `Joint::ResolveChildLinkX()`).

scpeters, 2020-08-14:
> For element queries, e.g. `LinkByName` vs. `LinkByNestedName`, `LinkByName` being able to resolve with "::" may be the best for backwards compatibility (the legacy composition behavior for include) as well as being able to easily resolve elements (e.g. `Joint::ResolveChildLinkX()`).

I reverted the new APIs in my prototype branch in https://github.com/osrf/sdformat/commit/bd618e97da4cd0f961bd49352691de70ecaea718 and added `::` syntax to the existing APIs instead.

EricCousineau-TRI, 2020-08-20:
@azeey is looking into the global graph.

EricCousineau-TRI, 2020-09-03:
Per f2f, summary on global graph is to couple the scoping and encapsulation to call sites (e.g. at each query to large graph from elements) -- vs. having some special aliasing view.

EricCousineau-TRI, 2020-09-24:
Per f2f, global graph w/ tests working.

Currently working on how to validate graphs (e.g. handling the no-longer-opaque nested models)
Has scoped graph (via `weak_ptr`), adds some scoping context.

PR #381 (2020-11-18): Allow "drilling down" into nested model frames
This a draft PR that adds functionality to allow drilling down into nested models with the `::` syntax. This also makes included models work as nested models

Resolves #293, #284
This builds on top of #355

---
This change is [[image]](https://reviewable.io/reviews/osrf/sdformat/381)
closed 2020-11-18 by azeey