gazebosim/sdformat #294: Model::Load should not emit duplicate child element warning for <plugin> elements

azeey, 2020-06-01:
`<plugin>` elements should be ignored when validating a model for name uniqueness, but `Model::Load` emits a warning if it encounters multiple `<plugin>` elements with the same name attribute.

Related: ignitionrobotics/ign-gazebo#164

scpeters, 2020-06-02:
In my opinion, `ign-gazebo` is misusing the `//plugin/@name` attribute by storing a non-unique class name there. I think it would be cleaner to use a custom attribute for the class name.

In general I think the parser shouldn't bother about things like `//plugin/@name`, but this attribute is in the spec saying that it [must be unique](https://github.com/osrf/sdformat/blob/master/sdf/1.7/plugin.sdf#L4-L5). So if we're going to change the parser to eliminate this warning, then I think we would also need to change the spec.

Another option is for ign-gazebo to start using a custom element for plugins, like `<ign-gazebo:system>` or something.

azeey, 2021-09-30:
Fixed by #721

PR #721 (2021-09-30): Make exception for plugins when checking for name uniqueness
# 🦟 Bug fix

Fixes #712 
Fixes #294

## Summary
As discussed in #712, #294 and possibly other issues, the `name` attribute of `<plugin>` is used by downstream applications like ign-gazebo to specify the class name within the plugin, so name collisions can occur frequently. This PR adds an exception for the `<plugin>` so that no warning is emitted when a plugin name matches the names of other entities.

I've added overloads to functions in  `sdf::Element` so that this PR can be backported to `sdf9`.

## Checklist
- [x] Signed all comm [... cut]
closed 2021-09-30 by azeey