gazebosim/sdformat #313: Precision loss when adding a nested model

hauke76, 2020-07-06:
Hi,

I stumbled over the issue that adding nested models results in a floating point precision loss. When adding a nested model, the `_includeSDF` is [converted to a string](https://github.com/osrf/sdformat/blob/master/src/parser.cc#L1325) through a call to `Element::ToString`. This ultimately calls `Param::GetAsString()` which uses the [ParamStreamer](https://github.com/osrf/sdformat/blob/master/include/sdf/Param.hh#L78) to write data to a `std::ostream`. By injecting `std::setprecision(std::numeric_limits<T>::max_digits10)` in the output stream this precision loss can be prevented.

Kind regards,
Hauke

hauke76, 2020-07-06:
I created a pull request for the master branch here: https://github.com/osrf/sdformat/pull/314
I have a [similar fix](https://github.com/osrf/sdformat/compare/sdf6...hauke76:fix_precision_loss_v6) available for v6.x and it would be great if you could consider that as a merge request as well.

EricCousineau-TRI, 2021-04-27:
Given merge of #314, closing this issue. Thanks!

closed 2021-04-27 by EricCousineau-TRI