gazebosim/sdformat #322: Whitespace preservation inconsistency

chapulina, 2020-07-29:
While backporting some tests from Dome to Blueprint (https://github.com/ignitionrobotics/ign-sensors/pull/33/), I noticed that there's an inconsistency with how `sdf8` and `sdf10` treat multiple whitespaces inside elements (not inside arguments). The following test fails on `sdf8` but passes on `sdf10`.

I believe this difference comes from the default behaviour of `tinyxml` versus `tinyxml2`, according to `tinyxml` documentation on [whitespace preservation](https://leethomason.github.io/tinyxml2/):  "_By default, TinyXML-2 preserves white space in a (hopefully) sane way that is almost compliant with the spec. (TinyXML-1 used a completely different model, much more similar to 'collapse'_".

It would be nice to be consistent across SDF versions. I can see arguments for each behaviour though. This is not very important for our use cases right now, but I thought we should keep track of it.

### Test

```.diff
diff --git a/src/SDF_TEST.cc b/src/SDF_TEST.cc
index 27f194ad..cca71d2e 100644
--- a/src/SDF_TEST.cc
+++ b/src/SDF_TEST.cc
@@ -696,6 +696,37 @@ TEST(SDF, WriteURIPath)
 }
 #endif  // _WIN32
 
+/////////////////////////////////////////////////
+TEST(SDF, readString)
+{
+  std::string testWorld = R"sdf(
+    <?xml version="1.0" ?>
+    <sdf version="1.5">
+      <world name="hello  world(123)">
+        <frame>
+          <name>hello  world(456)</name>
+        </frame>
+      </world>
+    </sdf>)sdf";
+
+  sdf::SDF sdfParsed;
+  sdf::initFile("root.sdf", sdfParsed.Root());
+  auto rootOriginal = sdfParsed.Root();
+  EXPECT_TRUE(sdf::readString(testWorld, rootOriginal));
+
+  EXPECT_TRUE(sdfParsed.Root()->HasElement("world"));
+  auto worldElem = sdfParsed.Root()->GetElement("world");
+  ASSERT_NE(nullptr, worldElem);
+
+  EXPECT_EQ("hello  world(123)", worldElem->Get<std::string>("name"));
+
+  EXPECT_TRUE(worldElem->HasElement("world"));
+  auto frameElem = worldElem->GetElement("frame");
+  ASSERT_NE(nullptr, frameElem);
+
+  EXPECT_EQ("hello  world(456)", frameElem->Get<std::string>("name"));
+}
+
 /////////////////////////////////////////////////
```

### Failure on `sdf8`

```
/home/chapulina/blueprint_ws/src/sdformat/src/SDF_TEST.cc:723: Failure
Value of: worldElem->HasElement("world")
  Actual: false
Expected: true
/home/chapulina/blueprint_ws/src/sdformat/src/SDF_TEST.cc:727: Failure
Expected equality of these values:
  "hello  world(456)"
  frameElem->Get<std::string>("name")
    Which is: "hello world(456)"
```

chapulina, 2020-08-20:
https://github.com/ignitionrobotics/ign-gazebo/issues/280 has a more important use case. I think it would be good to support this kind of declaration on SDF10:

```
<uri>
   https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Coke
</uri>
```

scpeters, 2020-08-31:
cc @mjcarroll

chapulina, 2020-09-10:
Fixed in #359

PR #359 (2020-09-10): Fix whitespace preservation behavior with TinyXML2
This addresses a change in how whitespace is preserved/collapsed between
tinyxml and tinyxml2. The change primarily impacts SDF files that have
newlines, which is frequently done for aesthetics with include uris.

In this case, we use an alternate constructor for TinyXML2 that will
collapse whitespace by default.

Note that there is a performance impact of this behavior, which causes
the parsing to run essentially twice.

More information on the behavior may be found here: https://leethomason.github.io/tinyxml2/

Closes #322
commit 784eb197 (2020-09-10): Fix whitespace preservation behavior with TinyXML2 (#359)

This addresses a change in how whitespace is preserved/collapsed between
tinyxml and tinyxml2. The change primarily impacts SDF files that have
newlines, which is frequently done for aesthetics with include uris.

In this case, we use an alternate constructor for TinyXML2 that will
collapse whitespace by default.

Note that there is a performance impact of this behavior, which causes
the parsing to run essentially twice.

More information on the behavior may be found here: https://leethomason.github.io/tinyxml2/

Closes #322