gazebosim/sdformat #323: ign.hh: Does not expose all functions consumed by `ign sdf`?

EricCousineau-TRI, 2020-07-29:
At present (f94626a889), `libsdformat` only exposes these functions in `ign.hh`:
https://github.com/osrf/sdformat/blob/f94626a8898b0a2898ae65892d76fd1031456133/src/ign.hh#L30-L37

However, `ign.cc` defines more (`cmdDescribe`, `cmdPrint`):
https://github.com/osrf/sdformat/blob/f94626a8898b0a2898ae65892d76fd1031456133/src/ign.cc

These are consumed by the intermediate Ruby script:
https://github.com/osrf/sdformat/blob/f94626a8898b0a2898ae65892d76fd1031456133/src/cmd/cmdsdformat.rb.in#L150-L158

It would be nice if these were exposed via the header file.

In https://github.com/robotlocomotion/drake/pull/13767, I want to use the `check` and `print` functionality to lint and convert models.

(I'm OK-ish with it being an unstable private API, but if it's there, would be nice to have it be consistent?)

jwnimmer-tri, 2022-05-24:
Now that cmdPrint is sprouting even more arguments, this issue has become even more acute.

```c++
extern "C" int cmdPrint(const char* _path, int _inDegrees, int _snapToDegrees,
                        float _snapTolerance, int _preserveIncludes,
                        int _outPrecision);
```

https://github.com/RobotLocomotion/drake/pull/17250

EricCousineau-TRI, 2022-05-24:
Good point! Alternative to exposing in header file is to simplify CLI interface (if possible), stripping need for Ruby.