gazebosim/sdformat #326: Joint type "continuous" and "revolute" appear redundant

sloretz, 2020-07-31:
SDFormat 1.6/1.7/1.8 XML has both a `revolute` and a `continuous` joint type, but they appear to be redundant. A `continuous` joint has no limits, while a revolute joint does. However, whether or not a joint has a limit is already given by the presence of a `<limit>` tag in the `<axis>` on the joint. Does `type="continuous"` mean a `<limit>` on the `<axis>` is an error? Similarly does `type="revolute"` mean the lack of an `<axis>` with a `<limit>` is an error?

It would seem that the `continuous` joint type could be removed if the spec was changed to say a joint of type `revolute` is limited if it has a `<limit>`, otherwise it's continuous.

https://github.com/osrf/sdformat/blob/f94626a8898b0a2898ae65892d76fd1031456133/sdf/1.7/joint.sdf#L11-L12

chapulina, 2020-08-01:
Interesting, I hadn't noticed that this was introduced to SDF. Neither Gazebo [nor Ignition](https://github.com/ignitionrobotics/ign-physics/blob/061965f69be7077c2ec3732ef4ef6b07f1a8cf06/dartsim/src/SDFFeatures.cc#L756-L758) currently support this joint type.

As a reference, [here's the PR where it was added](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/410/page/1), see [diff here](https://github.com/osrf/sdformat/compare/57f5f46891ba82f8087eb6e0265f6b8ab2ccec51..85a2735acd1f4c33624aa0f5633664f7d578d3ca).

> Does type="continuous" mean a <limit> on the <axis> is an error? Similarly does type="revolute" mean the lack of an <axis> with a <limit> is an error?

I think this documentation clarifies the intended semantics a bit:

https://github.com/osrf/sdformat/blob/f94626a8898b0a2898ae65892d76fd1031456133/include/sdf/JointAxis.hh#L162-L188

peci1, 2020-08-10:
Another thing that seems wrong about this is that the URDF->SDF converter thinks that SDF counterparts of URDF continuous joints should not have effort or velocity limits: https://github.com/osrf/sdformat/blob/913a7754c319190defa549bb73dc584e59f9734f/src/parser_urdf.cc#L3054 . I think the continuous/revolute should only be about the lower/upper positional limit.