gazebosim/sdformat #327: `//joint/axis/use_parent_model_frame` does not fail fast or issue a warning in SDFormat 1.7?

EricCousineau-TRI, 2020-08-03:
Ran across this here:
https://github.com/robotlocomotion/drake/pull/13758#pullrequestreview-459542067

Perhaps it did print out a console warning... will need to confirm.
Ideally, though, this would eventually cause things to fail fast. Perhaps in SDFormat 1.8, we can make custom elements much more strict? (e.g. it *must* be namespaced?)

FYI @scpeters @azeey

scpeters, 2020-08-04:
some history on this:

As of libsdformat 2.3.0, the parser would fail hard and print an `sdferr` console message if any unrecognized elements were parsed. This was relaxed to ignoring the unrecognized elements and printing an `sdfwarn` message in https://github.com/osrf/sdformat/commit/7999192c38f7b6508ce61820998a4d6c9401e437 ([bitbucket pull request 148](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/148/page/1)). The changed behavior was [released in libsdformat 2.3.1](https://github.com/osrf/sdformat/compare/sdformat2_2.3.0...sdformat2_2.3.1#diff-ba836a688e283ab18f0baee9ea2f7c38).

The behavior was changed further in https://github.com/osrf/sdformat/commit/1f9156feb28bbaa591465f0e837ac660eb51293e ([bitbucket pull request 449](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/449/page/1)) to stop ignoring unrecognized elements and instead copy them and print an `sdfdbg` message (which doesn't go to the console by default). This change was released in libsdformat 8.0.0.

We could consider moving back to either of these behaviors (ignore and warn, or fail and error) in `libsdformat10`.

azeey, 2020-08-05:
We'll probably have to check `OriginalVersion` to avoid failing on upconverted files that have unrecognized elements.

chapulina, 2020-10-13:
> We could consider moving back to either of these behaviors

Not to feature creep, but another option would be to make the behavior configurable :innocent: I can imagine even the same application using different behaviors at different times. For example, you may want to be more forgiving when loading an entire world file, but more strict when inserting a single model.

brawner, 2021-01-30:
Opened https://github.com/osrf/sdformat/pull/481 to add a configurable way to fail with unrecognized elements. It's more general than this specific use case, but thanks to @azeey ParserConfig PR (https://github.com/osrf/sdformat/pull/439) more customization is always possible.

EricCousineau-TRI, 2021-02-11:
Will do a quick PR to see if #481 addressed this issue.

@chapulina Does Steven's work address your request for configuration? (I assume so, but just wanna check!)

EricCousineau-TRI, 2021-02-11:
Confirmed and encoded in #490

PR #490 (2021-02-12): tests: Ensure removed SDFormat elements raise errors in newer versions
Resolves #327 (explicitly)
Follow-up to #481

Uses model setup from issue:
https://github.com/RobotLocomotion/drake/blob/ab20bf25d9e9c05f861d7b6905cd6ef4dd9950c7/examples/multibody/four_bar/four_bar.sdf#L249

\cc @brawner
closed 2021-02-12 by scpeters