gazebosim/sdformat #334: Option to disable sdformat.log

jwnimmer-tri, 2020-08-10:
See https://github.com/RobotLocomotion/drake/issues/9087.

Drake incorporates libsdformat as part of its stack.  Drake's policy as a library is that it should not autonomously create log files, but rather just offer the `spdlog` API for users to set up their own logging choices in their own application.  Drake's default needs to be "Do not write files in $HOME".

As such, it would be convenient for us if libsdformat offered an option to disable writing the `sdformat.log` file.

jwnimmer-tri, 2020-08-10:
In the meantime, I am using a work-around at https://github.com/RobotLocomotion/drake/pull/13850.

scpeters, 2020-08-31:
what mechanism would you prefer for enabling this option?

* wrap it in an `ifdef` so it could be disabled when compiling?
* read from an environment variable to determine whether to write to log file?

mjcarroll, 2020-08-31:
I think in addition to enabling/disabling internal SDFormat logging, there should additionally be a way of setting the log file destination.  I recently ran into this with Bazel, which doesn't like you to write outside of the file structure it creates.

Maybe something like:

```
SDFORMAT_LOG_DIRECTORY=$HOME
SDFORMAT_LOG_DISABLE=True
```

jwnimmer-tri, 2020-08-31:
The first option (`#ifdef`) would be slightly more ergonomic for us, but really anything would work.

Another alternative would be an API we could call to change the behavior.  This gets at another fundamental question of libsbformat that parsing should always be done against a parsing context or configuration, so that using-code can pass down customizations to any given request to parse, without having process-wide globals.  We could add more argument to `LoadSdfString` but that gets awkward when there are too many.  A parsing object would help consolidate (and reuse) the configuration.

In the long term, the best outcome is that libsdformat posts its messages (calls a function) on some logger interface or infrastructure, so that downstream code can easily capture / prefix / redirect / threshold the messages.  That might even be better formulated as a "parsing error" interface, rather than unstructured text logging.  A parsing library really has no business just remarking plain text into a console -- if the input text is malformed, then libsdformat should be reporting those errors in a structured way (with input line and column numbers, severity, configuration of strict vs permissive parsing, etc.) so that application code can inform the user in a way consistent with the application's behavior.

scpeters, 2020-09-01:
> The first option (`#ifdef`) would be slightly more ergonomic for us, but really anything would work.

I think an `#ifdef` would be the easiest thing to implement in the short term.

> Another alternative would be an API we could call to change the behavior. This gets at another fundamental question of libsbformat that parsing should always be done against a parsing context or [...]

This makes sense to me, but I don't think we'll have time to implement this or the other long-term suggestion in `libsdformat10`, which freezes next week.

scpeters, 2020-09-02:
I've proposed adding an `#ifdef` in #348

scpeters, 2020-09-07:
> The first option (`#ifdef`) would be slightly more ergonomic for us, but really anything would work.
>
> Another alternative would be an API we could call to change the behavior. This gets at another [...]

@jwnimmer-tri we've added the `#ifdef` option in #348. Should we close this issue or do you want it to pivot towards some of these other changes that you've mentioned here?

jwnimmer-tri, 2020-09-10:
I think we can open a new issue for the other future work.  I think the gist of it would be "Remove all global variables", e.g., `g_uriPathMap` as well as the console config, etc.

closed 2020-09-10 by jwnimmer-tri

scpeters, 2020-09-10:
We just hit feature freeze for `libsdformat10` and are aiming for a `libsdformat11` release in December or so. Paging @EricCousineau-TRI to see if we want to prioritize removing global variables for `libsdformat11`. If so, then we'll make an issue for it and add it to the relevant milestone.

EricCousineau-TRI, 2020-09-10:
Yes, removing globals (as a stretch) would be great. I would like the pose proposal to have priority (I favor features over software-that-still-works-but-needs-factoring when possible).

Can you make the issue?

scpeters, 2020-09-10:
> Yes, removing globals (as a stretch) would be great. I would like the pose proposal to have priority (I favor features over software-that-still-works-but-needs-factoring when possible).
> 
> Can you make the issue?

#373