gazebosim/sdformat #336: SDFormat 1.8: Should permit nesting models without links (e.g. defining frames / visuals / collision only)?

EricCousineau-TRI, 2020-08-13:
In SDFormat 1.7, we admit models with no links iff they are `//static=true`, and the canonical link for those models are attached to the world:
http://sdformat.org/tutorials?tut=pose_frame_semantics_proposal&cat=pose_semantics_docs&#2-1-implicit-frame-defined-by-model-pose-attached-to-canonical-link

However, @azeey pointed out that we may want similar behavior for nested models:
https://github.com/osrf/sdf_tutorials/pull/34#pullrequestreview-466335947

EricCousineau-TRI, 2020-08-13:
Thoughts:
- Not sure how I personally feel about just adding frames / visuals / collision without inertias. (if you're adding new collisions via SDFormat, they should be real and physical?)
- Not sure how I feel about this when the adding element will have scoped names.

So thinking about this, my gut says we punt for now on SDFormat 1.8.

EricCousineau-TRI, 2020-12-17:
This should've been accomplished for static models per latest PRs from @azeey 

@brawner to confirm that this is already covered in a unittest.

brawner, 2020-12-17:
Resolved via https://github.com/osrf/sdformat/pull/341 and is tested in the `model_dom.cc` integration test `DOMRoot.LoadNestedCanonicalLink` and `DOMRoot.LoadNestedExplicitCanonicalLink` (https://github.com/osrf/sdformat/blob/5a910816ef57ebd34aa4cf0c0a233750d8af88b3/test/integration/model_dom.cc#L433-L531)

closed 2020-12-17 by brawner