gazebosim/sdformat #351: Consider using Ignition Common's Console

chapulina, 2020-09-02:
libSDFormat's [Console](https://github.com/osrf/sdformat/blob/master/include/sdf/Console.hh) class provides a lot of the same functionality as [ignition::common::Console](https://github.com/ignitionrobotics/ign-common/blob/master/include/ignition/common/Console.hh).

The same way that SDF's custom math types were deprecated in favor of Ignition Math types (https://github.com/osrf/sdformat/issues/91), I think it could be beneficial to deprecate SDF's custom console to reduce duplication from Ignition. Ignition Common's core component shouldn't bring any new dependencies to SDFormat, I believe it's largest dependencies are `ign-math` and `tunyxml2`.

scpeters, 2020-10-18:
Component dependencies can be separated on Ubuntu easily enough, but it is not so easy on homebrew. Adding an `ign-common` dependency would add all ign-common component dependencies on macOS (including ffmpeg).

Another option is `console_bridge`

traversaro, 2020-10-19:
Indeed, for package managers that do not support optional dependencies or splitting the packages in fine grained packages, the ignition-common dependency can be problematic, even because given the complexity of the SDF spec, ideally sdformat should be a small dependency that can be integrated with low friction (ideally with CMake's [`FetchContent`](https://cmake.org/cmake/help/v3.11/module/FetchContent.html)) in a lot of projects, even embedded ones.

chapulina, 2020-10-20:
> Adding an ign-common dependency would add all ign-common component dependencies on macOS

Ouch yeah this makes the split of `ign-common` into components (https://github.com/ignitionrobotics/ign-common/issues/14) not really useful.

scpeters, 2020-10-21:
> > Adding an ign-common dependency would add all ign-common component dependencies on macOS
> 
> Ouch yeah this makes the split of `ign-common` into components ([ignitionrobotics/ign-common#14](https://github.com/ignitionrobotics/ign-common/issues/14)) not really useful.

it's very useful on some platforms; it's just not universally useful due to the limitations of some packaging systems

chapulina, 2020-10-21:
> it's very useful on some platforms;

Sure, but it needs to be useful on all our supported platforms, otherwise we lose the benefit, like in this situation. `sdformat` isn't the only library in need of a console revamp. `ign-transport` for example is filled with `std::cout`s that aren't logged together with the other libraries.

> it is not so easy on homebrew

Would it be possible to add separate formulae for each `ign-common` component, and leave the `ign-common3` formula as a "meta-formula" that just brings the rest? If this isn't a crazy idea, I can open an issue on `homebrew-simulation` to stop hijacking this issue.

chapulina, 2020-11-03:
Ticketed an issue about splitting the homebrew formulae: https://github.com/osrf/homebrew-simulation/issues/1196