gazebosim/sdformat #367: Release version 9.3.0 into Ubuntu 20.10

scpeters, 2020-09-07:
We have made significant changes on the `sdf9` branch recently to support nested models in SDFormat 1.7 (#283, #341) and are planning to release it very soon in version 9.3.0. We think this will be ok, since sdformat hasn't yet had a version 9 release in Ubuntu, though it looks like 9.2.0 will be in 20.10:

* https://packages.ubuntu.com/groovy/libsdformat9-9

It would be really helpful for the first version of libsdformat9 to be included in Ubuntu to be version 9.3.0. We plan to release it this week.

cc @j-rivero

scpeters, 2020-09-07:
osrf/gazebo support for this version of libsdformat is added in https://github.com/osrf/gazebo/pull/2824

do we need to make a release of that as well for this to be accepted in Ubuntu?

j-rivero, 2020-09-07:
> It would be really helpful for the first version of libsdformat9 to be included in Ubuntu to be version 9.3.0. We plan to release it this week.

The Ubuntu auto freeze process is already finished but I can try to ask for an update if you consider it important.

> do we need to make a release of that as well for this to be accepted in Ubuntu?

Yes. That would be the first step.

scpeters, 2020-09-08:
the 9.3.0 release has been made:

* https://github.com/osrf/sdformat/releases/tag/sdformat9_9.3.0

scpeters, 2020-09-22:
@j-rivero is there a status page I should watch to check the status of this package?

j-rivero, 2020-10-12:
Ubuntu FFe tracker https://bugs.launchpad.net/ubuntu/+source/sdformat/+bug/1895553

j-rivero, 2020-11-16:
Unfortunately it was not able to make it for 20.04. I'll update sdformat on Debian to be sure we catch the train for 21.04.

closed 2020-11-16 by j-rivero