gazebosim/sdformat #373: Stop using global variables

scpeters, 2020-09-10:
There are several places where global variables are used in `libsdformat`, which makes it difficult to control behavior when there are multiple paths that parse SDFormat files. As @jwnimmer-tri noted in https://github.com/osrf/sdformat/issues/334#issuecomment-683985626:

> parsing should always be done against a parsing context or configuration, so that using-code can pass down customizations to any given request to parse, without having process-wide globals. We could add more argument to `LoadSdfString` but that gets awkward when there are too many. A parsing object would help consolidate (and reuse) the configuration.

> In the long term, the best outcome is that libsdformat posts its messages (calls a function) on some logger interface or infrastructure, so that downstream code can easily capture / prefix / redirect / threshold the messages. That might even be better formulated as a "parsing error" interface, rather than unstructured text logging. A parsing library really has no business just remarking plain text into a console -- if the input text is malformed, then libsdformat should be reporting those errors in a structured way (with input line and column numbers, severity, configuration of strict vs permissive parsing, etc.) so that application code can inform the user in a way consistent with the application's behavior.

### Places where global variables are used:

* https://github.com/osrf/sdformat/blob/sdformat10_10.0.0-pre2/src/Console.cc#L31-L42
* https://github.com/osrf/sdformat/blob/sdformat10_10.0.0-pre2/src/SDF.cc#L43-L47
* https://github.com/osrf/sdformat/blob/sdformat10_10.0.0-pre2/src/parser_urdf.cc#L53-L63

EricCousineau-TRI, 2021-02-11:
@azeey Do you think we should close this out, or keep it open to simplify the number of overloads in the future (via a tick-tock)?

azeey, 2021-02-25:
I vote for closing this out and creating a new issue for addressing the number of overloads.

EricCousineau-TRI, 2021-02-25:
Closed via #439

jwnimmer-tri, 2021-02-25:
Here are some globals still found on the head of master.  Maybe I checked the wrong branch?

https://github.com/osrf/sdformat/blob/f59a342469bd9c71bd2dc5b325d777164375b06a/src/SDF.cc#L40
https://github.com/osrf/sdformat/blob/f59a342469bd9c71bd2dc5b325d777164375b06a/src/parser_urdf.cc#L52-L63
https://github.com/osrf/sdformat/blob/f59a342469bd9c71bd2dc5b325d777164375b06a/src/Console.cc#L43

jwnimmer-tri, 2021-02-25:
Possibly the "configuration difficulties" issue is resolved, but in general library code should avoid using globals as much as possible, both for the usual [style lguide](https://google.github.io/styleguide/cppguide.html#Static_and_Global_Variables) and [fiasco](https://en.cppreference.com/w/cpp/language/siof) reasons, and also in support of multi-threaded applications (avoiding race conditions).

There are rare cases where it's impractical to avoid globals, but the links I posted above are all easily fixable with modern C++.