gazebosim/sdformat #375: Ampersands not showing in generated SDF collide bitmask doc

PeterMitrano, 2020-09-18:
The documentation source says
```Collision happens if ((category1 & collision2) | (category2 & collision1)) is not zero.```

Unfortunately the ampersands get lost in generating the html, and the website shows this, which is quite confusing

[image]

EricCousineau-TRI, 2020-12-10:
@claireyywang Might you have time to look into this by Jan. 21?

claireyywang, 2020-12-10:
@EricCousineau-TRI yeah sure I'll take a look.

PeterMitrano, 2020-12-10:
Nice to see TRI investing in these open source projects! Thanks from UMich :)

EricCousineau-TRI, 2021-01-14:
Should run `xmllint` on the corresponding schema `*.sdf` files. If the `&` exists directly in the files, they should use the `&amp;` escape character.

azeey, 2021-02-04:
Fixed by #483

PR #483 (2021-02-04): Wrap description tags in CDATA
closes #375

This PR wraps all `<description>` content in `<![CDATA[]]>`, which was stripped away when full sdf files were generated. All symbols should show up properly on website upon merging.
closed 2021-02-04 by azeey