gazebosim/sdformat #380: Inertial `<pose relative_to="some_frame">` has no effect

sloretz, 2020-09-29:
I'm unable to get the pose of an `<inertial>` relative to a frame using `sdformat9`. It looks like the `relative_to` attribute has no effect if used in the `<pose>` of an `<inertial>`. The pose is always treated as if it were in link frame.

[image]

```xml
<?xml version='1.0' encoding='utf-8'?>
<sdf version="1.7">
  <model name="pose_inertial_in_frame">
    <frame name="some_frame">
      <pose>0 0 1 0 0 0</pose>-->
    </frame>
    <link name="pose_inertial_in_frame_link">
      <visual name="pose_inertial_in_frame_link_visual">
        <pose relative_to="some_frame">0 0 0 0 0 0</pose>
        <geometry>
          <box>
            <size>0.1 0.2 0.4</size>
          </box>
        </geometry>
      </visual>
      <collision name="pose_inertial_in_frame_link_collision">
        <pose relative_to="some_frame">0 0 0 0 0 0</pose>
        <geometry>
          <box>
            <size>0.1 0.2 0.4</size>
          </box>
        </geometry>
      </collision>
      <inertial>
        <pose relative_to="some_frame">0 0 0 0 0 0</pose>
        <mass>1.23</mass>
        <inertia>
          <ixx>0.0205</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.017425</iyy>
          <iyz>0</iyz>
          <izz>0.005125</izz>
        </inertia>
      </inertial>
    </link>
  </model>
</sdf>
```

It looks like the inertial is in the wrong pose using gazebo 11.1.0. It should be at the same location as the visual and collision, but it's in the ground.

The SDFormat API doesn't have a `SemanticPose()` API for inertials. I was expecting the [`Inertial()` API](http://osrf-distributions.s3.amazonaws.com/sdformat/api/9.0.0/classsdf_1_1v9_1_1Link.html#a0726c0fbfff0e926abda0f6fd8c9b401) to return a pose in link frame, but with the `relative_to` frame resolved. It looks like the issue is a variable `inertialFrame` is initialized, but never set with the true value:

https://github.com/osrf/sdformat/blob/b6c761ba38652ea440c4f728d60ef4b85cebf511/src/Link.cc#L169-L177

EricCousineau-TRI, 2020-12-17:
This is feasible for `libsdformat11` milestone:

- Add `Link::ResolveInertial(relative_to="")`
- Deprecate `Link::Inertial()`
- Replace with explicit `Link::RawInertial()`

Alternative:

- Fail fast if `//inertial/pose/relative_to` is non-default

EricCousineau-TRI, 2021-01-28:
Per f2f, would be simplest to error out if `//inertial/pose/@relative_to` is non-default.

It's probably a niche case to have that and `//link/pose/@relative_to` differ (creds to @sammy-tri)

closed 2021-02-03 by azeey