gazebosim/sdformat #391: Command ign sdf --check fails with fuel models

joxoby, 2020-10-15:
For example, running:
```bash
ign sdf -k ~/ignition_workspace/src/ign-gazebo/examples/worlds/multicopter_velocity_control.sdf
```

fails with:
```bash
Error [SDF.cc:165] Tried to use callback in sdf::findFile(), but the callback is empty.  Did you call sdf::setFindCallback()?Error [SDF.cc:165] Tried to use callback in sdf::findFile(), but the callback is empty.  Did you call sdf::setFindCallback()?Error: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/X3 UAV/4]
Error: Invalid uri[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/X3 UAV/4]. Should be model://https://fuel.ignitionrobotics.org/1.0/openrobotics/models/X3 UAV/4
Error: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/X4 UAV Config 1]
Error: Invalid uri[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/X4 UAV Config 1]. Should be model://https://fuel.ignitionrobotics.org/1.0/openrobotics/models/X4 UAV Config 1
```

chapulina, 2020-10-15:
`libsdformat` doesn't know about Fuel, so it's expected that it can't find the model.

This makes the `ign sdf --check` command line tool less useful though. `libsdformat` supports the addition of custom `findFile` callbacks, so I think it could handle it more gracefully when it can't find a resource.

I think it would be nice if the `ign sdf --check` call wouldn't stop and fail when it encounters an URI that it can't handle, and instead would continue running and just print a warning like "Warning: URI [] couldn't be resolved. If that's an SDF file, its contents can't be checked."

scpeters, 2020-10-19:
I think it is useful to have the `ign sdf` command in `libsdformat`, and we are not planning to give it a dependency on ignition fuel, so `ign sdf` wouldn't be able to support this feature.

On the other hand, it would be a useful feature, so I would suggest adding it as a sub-command of a package that uses both `libsdformat` and `ignition-fuel-tools`, such as ign-gazebo. I'm going to close this, but feel free to comment here if you open an issue or pull request to ign-gazebo about this.

closed 2020-10-19 by scpeters

joxoby, 2020-10-19:
> I think it is useful to have the ign sdf command in libsdformat, and we are not planning to give it a dependency on ignition fuel, so ign sdf wouldn't be able to support this feature.

Right, I totally agree.  However, the approach suggested by @chapulina would not add any dependencies. Is this something we would like to move forward with?