gazebosim/sdformat #401: Should store an included model's source file name

azeey, 2020-10-30:
This can be useful when a world is saved. Relates to a TODO in https://github.com/osrf/sdformat/blob/144f82cf8b994abd2a566a7bbe6ae873a9d92403/src/parser.cc#L1075-L1082

EricCousineau-TRI, 2020-10-30:
Can you illustrate this workflow, or point to a Gazebo tutorial / unittest that illustrates the nested model filenames being kept?

azeey, 2021-03-11:
I don't think there is currently an example of this, but when saving worlds in ign-gazebo, there is an option to preserve the `<include>` tag as opposed to expanding it to a `<model>`. This worked by making use of `sdf::Element::FilePath` of the included model (https://github.com/ignitionrobotics/ign-gazebo/blob/5348afa4b5ae4349f72aca51af01db856ae65b90/src/SdfGenerator.cc#L101-L124). This was somewhat of a workaround for the lack of information regarding what file the model was loaded from. It is also unable to retrieve any other information included in the original `<include>` element, such as `<plugin>` or any custom elements.

#509 addresses this by storing the original content of the `<include>` element so it can be retrieved at a later time for such purposes.

EricCousineau-TRI, 2021-03-11:
Sweet! Thanks!

PR #509 (2021-03-23): Store include info in Element
# 🎉 New feature

Closes #401 

## Summary
This stores the contents of `<include>` inside the Element loaded from the included file.
For example, given the following SDFormat:
```xml
<sdf version = '1.8'>
  <world name='default'>
    <include>
      <uri>model_uri</uri>
      <pose>1 2 3 0 0 0</pose>
    </include>
  </world>
</sdf>
```

The `ElementPtr` associated with the model loaded from 'model_uri' will
have its `includeElement` set to

```xml
    <include>
      <uri>model_uri</uri>
      <pose>1 2 3 0 0 0</pose>
    </include>
```
This can be used to retriev [... cut]
closed 2021-03-23 by azeey