gazebosim/sdformat #407: PoseNoValue test result is flaky on non-x64 architectures

traversaro, 2020-10-31:
There have been several reports that the PoseNoValue test seems to be flaky or to fail on non-x64 architectures, see for example:
* https://github.com/conda-forge/libsdformat-feedstock/pull/16#issuecomment-703321154

Related issue: https://github.com/osrf/sdformat/issues/228 .

scpeters, 2020-12-02:
this test is in [Utils_TEST.cc](https://github.com/osrf/sdformat/blob/sdformat10_10.0.0/src/Utils_TEST.cc#L56)

possible fix in https://github.com/ignitionrobotics/ign-math/pull/180

scpeters, 2020-12-02:
this has been failing consistently on i386:

* https://build.osrfoundation.org/job/sdformat-ci-master-bionic-i386/19/testReport/(root)/DOMUtils/PoseNoValue/

scpeters, 2020-12-02:
testing with https://github.com/ignitionrobotics/ign-math/pull/180 via https://github.com/ignition-tooling/release-tools/commit/b31dbf91cb14a8d59cfdd2e59f605eb269f313d6

[image] https://build.osrfoundation.org/job/sdformat-ci-master-bionic-i386/20/

scpeters, 2020-12-02:
this will be fixed after https://github.com/ignitionrobotics/ign-math/pull/180 is released in 6.7.1