gazebosim/sdformat #421: Document that force-torque sensor should only be attached to joints, not links

scpeters, 2020-11-19:
It has always been my understanding based on the [usage in osrf/gazebo](https://github.com/osrf/gazebo/blob/6fd426b3949c4ca73fa126cde68f5cc4a59522eb/test/worlds/force_torque_frame_test.world#L88-L163) that a sensor of type `force_torque` may only be a `//joint/sensor` and not a `//link/sensor`. I believe it is not stated in the specification, and the [test/sdf/sensors.sdf example](https://github.com/osrf/sdformat/blob/sdformat10_10.0.0/test/sdf/sensors.sdf#L140-L142) even includes a `//link/sensor` of type `force_torque`. I noted this while reviewing the new `sdf::ForceTorque` DOM element, which does not enforce this condition, in keeping with the pattern currently set.

I would propose that we formalize this in SDFormat 1.8 and generate parsing errors if a `//link/sensor` has type `force_torque`.

scpeters, 2021-03-16:
I discussed this with @azeey today, and the user story about force-torque sensors could use some improvement, as there are use cases in which a force-torque sensor can be attached to an end effector and used like a contact-wrench sensor rather. This could be considered attaching it to a link rather than a joint from a user perspective.

There's not quite enough time to sort this out for the libsdformat11 feature freeze, but something quick that we could do for libsdformat11 would be to add sensor accessor API's to the `Joint` DOM class, since those are currently missing. This would allow `Joint` sensors to be accessed from the DOM.

traversaro, 2021-03-16:
> I discussed this with @azeey today, and the user story about force-torque sensors could use some improvement, as there are use cases in which a force-torque sensor can be attached to an end effector [...]

Perhaps I am missing something, but that use case is not covered by `contact` sensors in SDF?

scpeters, 2021-03-18:
> Perhaps I am missing something, but that use case is not covered by `contact` sensors in SDF?

that was my first thought as well, but that is informed by knowledge of gazebo's implementation details, and @azeey pointed out that end users may not recognize that distinction. I think there is a documentation / narrative issue to be resolved for end users

scpeters, 2024-01-30:
> There's not quite enough time to sort this out for the libsdformat11 feature freeze, but something quick that we could do for libsdformat11 would be to add sensor accessor API's to the `Joint` DOM [...]

this was done, starting with #393

scpeters, 2024-01-30:
I've started a prototype that warns if a ForceTorque sensor is attached to a link in https://github.com/gazebosim/sdformat/commit/3953e7f011c7bec289984529f477bdf996ef36ee