gazebosim/sdformat #425: Missing documentation for what concerns camera sensors frame

Nicogene, 2020-12-01:
I was unable to find information about the default direction of the optical axis of a generic camera sensor.

Without explicit information, I was assuming:
- z-axis goes along the optical axis (pointing outward)
- x-axis points rightward (wrt to the sensor's image plane)
- y-axis points downward.

(See [here](https://www.ros.org/reps/rep-0103.html#suffix-frames) for reference)

But looking at a simple example like [this one](http://gazebosim.org/tutorials?tut=camera_distortion&cat=sensors), in absence of any `<pose>` specification, the optical axis seems to be along x, since for Gazebo Z is up and X is in front then the camera plane is defined by  YZ  instead of XY as I was assuming.

Can you confirm that?

For the users, it would be ideal if the documentation could specify the documentation should specify on which axis the optical axis is oriented, and which axes define the image plane.

Related downstream issue: https://github.com/robotology/icub-models/issues/69 

cc @traversaro

traversaro, 2020-12-01:
An additional point that could be confusing for users is the following. It seems that indeed the optical axis in Gazebo is assumed to be aligned with the X direction of the frame defined by the composition of the `sensor` element `<pose>`'s and the `camera` element `pose`, meaning that the image plane is the `YZ` plane. However, the elements under the `lens` element seem to refer to the `XY` as the camera plane, see the `fx`, `fy`, `cx` and `cy` in http://sdformat.org/spec?ver=1.7&elem=sensor#lens_intrinsics .

traversaro, 2020-12-01:
Interestingly, the convention currently assumed in Gazebo for SDF's Camera frame seems to be not compatible with the Drake one (see https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1sensors_1_1_camera_info.html), for which instead the Z-axis is indeed the viewing direction. fyi @EricCousineau-TRI

EricCousineau-TRI, 2020-12-03:
Per f2f, @azeey can look into this, and potentially document the conventions used for the sensor frame (axes, computer vision vs. (right-handed computer graphics), and possibly with a suggested rpy offset to align w/ another convention).

traversaro, 2020-12-03:
Thanks a lot @EricCousineau-TRI and @azeey !

traversaro, 2020-12-03:
>  and possibly with a suggested rpy offset to align w/ another convention.