gazebosim/sdformat #440: joints with links from nested models fail to parse

pcdangio, 2020-12-15:
According to the [SDFormat documentation](http://sdformat.org/tutorials?tut=composition&cat=specification&#joints-and-nested-models), the proper way to create a joint with a link in a nested model is to scope the link with the nested model's name. However, loading an SDF with a joint such as this gives errors stating that the link can't be found. It appears that the parsing methods aren't able to handle the prefixed scope of the nested model's link.

I've tried the minimal SDF file below by loading it through `sdf::v9::Root::Load()`:

```
<?xml version="1.0" ?>
<sdf version="1.7">
    <model name="top_model">
        <link name="link_a"/>
        <model name="nested_model">
            <link name="link_b"/>
        </model>
        <joint name="top_joint" type="revolute">
            <parent>link_a</parent>
            <child>nested_model::link_b</child>
        </joint>
    </model>
</sdf>
```
and it yields the following error when loading/parsing in libsdformat:

```
Child link with name[nested_model::link_b] specified by joint with name[top_joint] not found in model with name[top_model].
```

This originates in parser.cc, which seems to completely ignore the fact that prefixed scoping exists:
https://github.com/osrf/sdformat/blob/5a910816ef57ebd34aa4cf0c0a233750d8af88b3/src/parser.cc#L1793-L1804

pcdangio, 2020-12-15:
**Update:**

This only appears to occur with the child link of a joint; the same parser.cc check referenced in the above post is not run on parent link names.

In my original example, if the nested model's link is swapped from the joint's child to the joint's parent, the SDF loads fine without error.

I'm wondering if the parser.cc lines posted above should be removed. [SDFormat documentation](http://sdformat.org/tutorials?tut=composition&cat=specification&#joints-and-nested-models) states that:

_"Note, however, that libsdformat does not currently check that nested model links referenced by joints are valid. This responsibility is left for downstream applications."_

scpeters, 2020-12-15:
We made some pretty extensive changes in SDFormat 1.7 in order to support the `//frame` and `//pose/@relative_to` data structures and didn't have quite enough time to support the `::` syntax. We have support for this syntax in SDFormat 1.8, which will be supported in `libsdformat11` (currently on the `master` branch), with the first bit of support added in #355 and then all the rest in #381.

azeey, 2020-12-15:
In the example you mentioned, the model is nested using `<include>`. In SDFormat 1.7, this is different from directly nesting the model using `<model>`. Referencing links and other entities that are children of a directly nested model is not supported in libsdformat9 and libsdformat10. However, due to the way `<include>` is [implemented](http://sdformat.org/tutorials?tut=composition&cat=specification&#libsdformats-implementation-of-include-in-models), referencing nested entities works if the model is nested using `<include>`.

> "Note, however, that libsdformat does not currently check that nested model links referenced by joints are valid. This responsibility is left for downstream applications."

I think this is still true. I believe the `checkJointParentChildLinkNames` function you linked to is only called when running `ign sdf -k`. It's not called when in `sdf::Root::Load`, which is what most downstream applications would use.

pcdangio, 2021-01-10:
Thanks for the additional information!

closed 2021-01-10 by pcdangio