gazebosim/sdformat #499: RawPose in world frame?

varunagrawal, 2021-02-19:
Thank you for all the great work with sdformat!

`sdformat9` deprecated `Pose()` in favor of `RawPose()`. While `Pose()` gave the link pose in the world frame, `RawPose()` gives the pose with respect to its relative frame (e.g. the joint it is connected to). This is causing challenges for us with respect to our test suite.

Is there an easy way to use the SDF model to get the pose in the world frame? That should be sufficient to make our code backwards compatible.

scpeters, 2021-02-19:
tl;dr look up the `SemanticPose::Resolve()` API

We have some documentation of the difference in how poses and relative frames are defined in the SDFormat specification between versions 1.6 and 1.7:

* documentation of [poses and relative frames in SDFormat 1.6 and earlier](http://sdformat.org/tutorials?tut=pose_frame_semantics&ver=1.5&cat=specification&)
* documentation of [poses and relative frames in SDFormat 1.7](http://sdformat.org/tutorials?tut=pose_frame_semantics&ver=1.7&cat=specification&)

Those pages are both focused on the SDFormat specification, but it appears that your question is about the C++ API. I think we don't currently have a tutorial about using the c++ API. Since determining the value of a `<pose>` element relative to a specific frame may require operations on a graph, we have provided the [SemanticPose::Resolve()](https://github.com/osrf/sdformat/blob/sdformat9_9.5.0/include/sdf/SemanticPose.hh#L65-L73) API. There is code in some of our tests that may be useful to see how the API can be used. For example, [this test](https://github.com/osrf/sdformat/blob/sdformat9_9.5.0/test/integration/model_dom.cc#L203-L274) uses the API to resolve relative poses of nested models in [this file](https://github.com/osrf/sdformat/blob/sdformat9_9.5.0/test/sdf/model_nested_model_relative_to.sdf).

varunagrawal, 2021-03-15:
Thanks @scpeters your suggestion worked.

For future reference to anyone who stumbles across this, the snippet that does it for me is:

```cpp
auto pose = link.RawPose();
auto errors = link.SemanticPose().Resolve(pose, ""); // get pose in world/base frame
// pose is now in the correct frame if errors.size() == 0.
```

closed 2021-03-15 by varunagrawal