gazebosim/sdformat #505: Specify max SDF version in URDF->SDF conversion

peci1, 2021-03-04:
## Desired behavior
`ign sdf -p` should accept a version parameter that would tell it the maximum SDF feature level it can use in the output

## Alternatives considered
Building an old version of sdformat from source... But that's not the solution.

## Additional context
I have a Xacro/URDF model which I would like to use both in Ignition Gazebo and Gazebo classic. But `gz sdf` doesn't understand some tags in the URDF, and `ign sdf` generates the file according to SDF 1.7 spec, which Gazebo 9 doesn't understand. So there's no way for me to use the model in both simulators.

peci1, 2021-03-04:
Also, I searched for 5 minutes, and haven't found any place that would clearly state what version of SDF spec is supported by which version of sdformat...

azeey, 2021-03-18:
This would be great to have, but I don't think it would be trivial because there is no mechanism for libsdformat to support multiple versions of the spec at the same time without upconverting to the latest spec it knows about.

> Also, I searched for 5 minutes, and haven't found any place that would clearly state what version of SDF spec is supported by which version of sdformat...

It's listed in http://sdformat.org/tutorials?tut=roadmap&cat=developers&. Maybe we should link back to this from the downloads page.