gazebosim/sdformat #545: sdformat: Website schemas are old and URLs are not versioned?

EricCousineau-TRI, 2021-04-22:
## Environment
N/A

## Description
* Expected behavior: Latest schemas are visible via a clearly versioned mechanism.
* Actual behavior: Current schemas are old (e.g. http://sdformat.org/schemas/root.xsd), and are not versioned. Also, all remote assets that have cross-references require remote fetching. May inline.

## Steps to reproduce
?

## Output
N/A

EricCousineau-TRI, 2021-05-20:
Per VC:
- `<?xml-model?>` is the mechanism used in `package.xml`: http://xml-model.info/
- Example usage of `<?xml-model..?>`: https://github.com/ros2/rcl/blob/3.1.2/rcl/package.xml#L2
- Should document that it's not necessary on SDFormat (given complexity), but can help given other tooling (e.g. editors and online linting / auto-complete)

FirefoxMetzger, 2021-07-23:
I ran into this issue today as well. In particular, not all files are present on the schema server. At the time of this writing, the following ones are missing:

- http://sdformat.org/schemas/frame.xsd
- http://sdformat.org/schemas/pose.xsd
- http://sdformat.org/schemas/atmosphere.xsd
- http://sdformat.org/schemas/particle_emitter.xsd
- http://sdformat.org/schemas/capsule_shape.xsd
- http://sdformat.org/schemas/ellipsoid_shape.xsd
- http://sdformat.org/schemas/link_state.xsd
- http://sdformat.org/schemas/air_pressure.xsd
- http://sdformat.org/schemas/lidar.xsd
- http://sdformat.org/schemas/model_state.xsd
- http://sdformat.org/schemas/light_state.xsd