gazebosim/sdformat #572: TestFile and SourceFile in test_config.hh appends paths without sanitization

aaronchongth, 2021-05-19:
## Environment
* OS Version: Ubuntu 20.04
* Source build, branch `sdf11`

## Description
* Expected behavior: 

When calling `sdf::testing::TestFile` on a model name that is a URI, for example,
```
<include>
  <uri>test_model/model.sdf</uri>
</include>
```
it should provide the absolute path like so, `/path/to/models/test_model/model.sdf`.

* Actual behavior: 

In windows machines however, we will produce a path like so `C:/somewhere\\path\\to\\models\\test_model/model.sdf`, due to `sdf::testing::TestFile` appending paths without sanitization.

azeey, 2021-05-20:
I thought the issue was actually in `sdf::findFile` when it's used to find the included file: https://github.com/osrf/sdformat/blob/e5ca000c1934b59302362ef85016ae20af501ed0/src/parser.cc#L1103-L1106

In your example, doesn't `sdf::Element::FilePath()` return `C:/somewhere\\path\\to\\models\\test_model/model.sdf`?

aaronchongth, 2021-05-21:
Yup, it did. Sorry I got confused myself for a moment. I just did some basic tests, and it is as you mentioned, it was `sdf::findFile` in `SDFImpl.hh`, https://github.com/osrf/sdformat/blob/sdf11/include/sdf/SDFImpl.hh#L82-L99 that does not sanitize the path.

I will make the changes to the issue accordingly, sorry for the confusion.