gazebosim/sdformat #579: URDF color is different between RVIZ and ign-gazebo

ahcorde, 2021-05-24:
[image]

Refer to this comment https://github.com/osrf/sdformat/pull/526#pullrequestreview-665798490

chapulina, 2021-06-17:
From @azeey's original comment:

> I tried to using an example to compare the colors between Rviz and ign-gazebo

Did you use Ogre 1 with Gazebo? The empty world defaults to Ogre 2

azeey, 2021-06-17:
Ah, I think I used the default so I was testing with Ogre2. With Ogre1 I was able to get a more similar result. I did have to reduce the intensity of the `sun` light though. I'd be okay with closing this ticket.

[image]

<details>
<summary>For reference, here are the world file and the model I used</summary>

```xml
<!-- empty.sdf -->
<sdf version="1.6">
  <world name="empty">

    <scene>
      <ambient>0.4 0.4 0.4</ambient>
      <background>0.8 0.8 0.8</background>
      <grid>true</grid>
    </scene>

    <gui fullscreen="0">

      <!-- 3D scene -->
      <plugin filename="GzScene3D" name="3D View">
        <engine>ogre</engine>
        <scene>scene</scene>
        <camera_pose>2 0 0.7 0 0.0 3.14</camera_pose>
      </plugin>
    </gui>

    <light type="directional" name="sun">
      <cast_shadows>false</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <intensity>0.5</intensity>
      <direction>-1 0 -1</direction>
    </light>
  </world>
</sdf>

```

```xml
<!-- test_material/model.urdf -->
<?xml version="1.0"?>
<robot name="test_robot">
  <link name="base_link">
    <visual>
      <geometry>
        <cylinder length="0.6" radius="0.2"/>
      </geometry>
      <material name="blue">
        <color rgba="0.5 0 .5 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <cylinder length="0.6" radius="0.2"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="10"/>
      <inertia ixx="0.4" ixy="0.0" ixz="0.0" iyy="0.4" iyz="0.0" izz="0.2"/>
    </inertial>
  </link>
</robot>
```
</details>

P.S I learned that `ogre2` doesn't use the values in `<scene>` while `ogre` does.

chapulina, 2021-06-17:
> I learned that ogre2 doesn't use the values in <scene> while ogre does.

Ouch, sounds like an `ign-gazebo` issue, mind ticketing it?

> I'd be okay with closing this ticket.

Cool, let's close it for now

closed 2021-06-17 by chapulina