gazebosim/sdformat #633: XSD schema generation doesn't parse nested includes (only top-level)

FirefoxMetzger, 2021-07-24:
The current `xmlschema.b` parses each `X.sdf` file and generates an accompanying `X.xsd` file; however, if any element other than the root element contains a nested `<include>` tag this tag is ignored. This is because the root element is parsed by the function `printXSD`, whereas other elements are parsed by `printElem` and the latter doesn't check for `<include>` tags.

This is a problem with more recent SDFormat files, which introduce `<pose>` as a complex type with an optionally nested `<relative_to>`. Pose is now `<include>`d, and hence never makes it into the produced `.xsd`.

The two solutions I can think of are: (1) copy the include parsing code from printXSD to printElem (works, but dirty). Refactor `xmlschema.b` and merge all code paths that parse element tags.

FirefoxMetzger, 2021-07-25:
The same is true for attributes that don't occur on the root level. An example is `imu.sdf` and its `gravity_dir_x`. The sdf specifies a `parent_frame` attribute, but this is missing from the generated xsd. It occurs any time when a non-root element is a complexType (i.e., its schema has sub-elements other than annotation).

slgrobotics, 2024-05-01:
@FirefoxMetzger @chapulina  - is there a chance that this bug will be fixed anytime soon? We would like to automatically check validity of SDF files against schema, and this looks impossible as the schemas are incomplete. Any advice would help. Thanks!  

(@dagar - FYI)