gazebosim/sdformat #639: environment_map not declared by material/pbr/specular in material.sdf

FirefoxMetzger, 2021-07-28:
The test sdf `material_pbr.sdf` uses an element called `environment_map` inside of the specular element of a pbr material:

https://github.com/ignitionrobotics/sdformat/blob/e4c658f14b73ab839d730c7a402a82a7dfa6a16e/test/sdf/material_pbr.sdf#L36

However, I can not find the matching tag on the website, and it is also missing from the `material.sdf`.

Website for 1.8: http://sdformat.org/spec?ver=1.8&elem=material#pbr_specular
Website for 1.6 (test uses this version): http://sdformat.org/spec?ver=1.6&elem=material#pbr_specular
Source Code: https://github.com/ignitionrobotics/sdformat/blob/e4c658f14b73ab839d730c7a402a82a7dfa6a16e/sdf/1.8/material.sdf#L112-L153

@jennuine Does the tag falsely declare the element, or is the element missing from the sdf?

FirefoxMetzger, 2021-07-29:
The same is true for this element

https://github.com/ignitionrobotics/sdformat/blob/e4c658f14b73ab839d730c7a402a82a7dfa6a16e/test/sdf/material_pbr.sdf#L39

inside the metal element of the same file. It is not documented in v1.8 or v1.6 (the version of the test file).

jennuine, 2021-08-18:
The sdf spec and website have been updated: http://sdformat.org/spec?ver=1.8&elem=material#specular_environment_map

closed 2021-08-18 by jennuine