gazebosim/sdformat #640: SDFormat 1.8 and 1.7 not backwards compatible with 1.6

FirefoxMetzger, 2021-07-28:
This is only an issue, if vX.X actually refers to semantic version (which it usually does, but it could be different here, and I didn't find it). If the SDFormat version tag doesn't imply semver, please close this issue :)

In `joint.sdf` the child element `use_parent_model_frame` was dropped in v1.7. This currently destroys (backward) compatibility of parsers of v1.7+ and SDF of v1.6 and below. My suggestion would be to re-introduce the element combined with a deprecation warning and/or a note in the documentation that this is deprecated and will be removed in version 2.0. Alternatively, the version of SDFormat could be bumped to v2.0, but I understand that there is general reluctance towards increasing version major.

Relevant docs:
http://sdformat.org/spec?ver=1.6&elem=joint#axis_use_parent_model_frame
http://sdformat.org/spec?ver=1.8&elem=joint#joint_axis

traversaro, 2021-07-28:
I vaguely recall that this was already discussed somewhere and indeed it was a deliberate choice that minor changes in SDF could result in non-backward compatible changes, but I can't remember where that discussion was.

traversaro, 2021-07-29:
I found it, the comment was by @scpeters and is in https://discourse.ros.org/t/sdformat-pose-frame-semantics-proposal/9852/6 .

azeey, 2021-07-29:
The SDFormat doesn't quite follow semver as it doesn't have patch versions. That being said, we do try to keep backward compatibility for users of the spec via a mechanism that automatically converts older versions to the latest (https://github.com/ignitionrobotics/sdformat/blob/sdf11/sdf/1.7/1_6.convert). This doesn't help parsers though, so yeah, I think our versioning scheme needs improvement.

FirefoxMetzger, 2021-07-29:
Thanks for that @traversaro and @azeey . Based on @traversaro 's comment I decided to treat each version as unique. Now I parse the SDF with a generic XML parser (SAX style) until I find the opening tag `<sdf version="X.X">`, read the version, and then load my SDF parser/schema for version X.X. This does make me think about the weird edge case where a `<sdf version="1.8">` includes a `<sdf version="1.6">` model or vice versa. I should check how the C++ parser in this repo deals with such a scenario...

Going back to the issue itself, your comment @azeey seems to solve the issue. If I understand it correctly, it's essentially a wontfix since SDFormat doesn't follow semver and doesn't (yet?) seem to implement a depreciation cycle. I'll close this issue, but feel free to reopen if I misunderstood :)

>  a mechanism that automatically converts older versions to the latest

That's an interesting comment @azeey . I'll have to look into how this works.

closed 2021-07-29 by FirefoxMetzger