gazebosim/sdformat #641: missing tag `extra` in include of elements

FirefoxMetzger, 2021-07-28:
I've noticed that some tests use an additional element inside the include element called `<extra>`, which is not documented on the website and which is missing from the respective `.sdf`'s. These seem to occur in test sdfs of the format `X_include_with_interface_api.sdf`. Is this an undocumented feature of SDF, something internal that should be ignored, or an old test?

Example: https://github.com/ignitionrobotics/sdformat/blob/sdf11/test/sdf/world_include_with_interface_api.sdf

FirefoxMetzger, 2021-07-28:
I also have the same question for the `<plugin>` tag occurring in here:
https://github.com/ignitionrobotics/sdformat/blob/e4c658f14b73ab839d730c7a402a82a7dfa6a16e/test/sdf/includes.sdf#L40

azeey, 2021-08-09:
The `X_include_with_interface_api.sdf` files are used for testing the SDFormat [Interface API](http://sdformat.org/tutorials?tut=composition_proposal&cat=pose_semantics_docs&#1-5-minimal-libsdformat-interface-types-for-non-sdformat-models). When using this API, any element in `//include` that's not used to override the included model are passed to the custom parser. However, since `<extra>` is not part of the spec, so it should really be `<custom:extra>`.

> I also have the same question for the <plugin> tag occurring in here:

I think that's an omission. `<plugin>` is allowed in `<include>`.

FirefoxMetzger, 2021-08-10:
> I think that's an omission. <plugin> is allowed in <include>.

In which version did this become allowed? Also, are there any other attributes that are allowed and should be overwritten?

azeey, 2021-08-10:
> In which version did this become allowed? Also, are there any other attributes that are allowed and should be overwritten?

I believe it's been allowed since the introduction of `<include>`, but I'm not 100% sure. The elements that can be overridden are listed in http://sdformat.org/tutorials?tut=composition&cat=specification&#including-a-model.

 > Does this mean that include should have the same line as plugin:

I don't think so because for `<include>` we should emit a warning if there are unknown elements whereas for `<plugin>` we allow any element.

FirefoxMetzger, 2021-08-12:
> I don't think so because for <include> we should emit a warning if there are unknown elements whereas for <plugin> we allow any element.

Hm, then I'm not quite sure how this is best solved. The schema essentially whitelists allowed elements, so if any element may show up (warning or not), then this needs to be specified. We could introduce a new element specifically for this, but I think it will be much easier to reuse the element used in //plugin and make the decision of emitting a warning an implementation detail of the parser.