gazebosim/sdformat #642: Invalid default value for elements of type <time>

FirefoxMetzger, 2021-07-28:
In `types.xsd` time is defined as a simple double, however inside `state.sdf` a string value of the form `"0 0"` is used as default. The documentation agrees with the latter and says it should be of the form `"{seconds} {nanoseconds}`.

I implemented a local fix inside `types.xsd` of the form

```xml
  <xsd:simpleType name="time">
    <xsd:restriction base="xsd:string">
      <xsd:pattern value="\d+ \d+"/>
    </xsd:restriction>
  </xsd:simpleType>
```

However, it still makes sense to bring it up as an issue here, since I don't know if this is the preferred solution.

chapulina, 2021-08-09:
The suggested fix makes sense to me

azeey, 2021-08-09:
I believe "{seconds} {nanoseconds}" is what's used by Gazebo classic, so the fix makes sense to me too.

PR #662 (2021-08-16): BUG: make time type string [s ns]
# 🦟 Bug fix

Fixes #642 

## Summary
While I did this in #643 I think it makes sense to make it a separate PR and rebase. This way this bug can be fixed while awaiting feedback for different things in the other PR.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contri [... cut]
closed 2021-08-16 by jennuine