gazebosim/sdformat #646: Implement snap-to feature for //pose/@rotation_type conversion

EricCousineau-TRI, 2021-07-29:
## Desired behavior
Per proposal http://sdformat.org/tutorials?tut=better_pose_proposal&cat=pose_semantics_docs&, should allow conversion among rotation types (e.g. from `rpy_degrees

## Alternatives considered
See proposal

## Implementation suggestion
See proposal

## Additional context
See proposal

FirefoxMetzger, 2021-07-30:
@EricCousineau-TRI Is it possible to comment on proposals somewhere? Based on the motivation, I was wondering if the following alternative has been considered:

```xml
<pose>
  <value degrees="True">0 0 0 90 45 0</value>
</pose>
```

Making it `degrees=False` by default would make the entire thing backward compatible with 1.8.

chapulina, 2021-07-30:
@FirefoxMetzger , those are hosted at `sdf_tutorials`, i.e. https://github.com/ignitionrobotics/sdf_tutorials/pull/37

FirefoxMetzger, 2021-07-30:
@chapulina Thanks :) Would I create an issue there to comment on a proposal, or what is the preferred approach?

chapulina, 2021-07-30:
Good question, I'd say it's best to open an issue here at `sdformat` for greater visibility. If you have specific suggestions, you could open a PR against the proposal.

azeey, 2021-07-30:
FWIW, the option we're currently working in #589 is `<pose rotation_type="rpy_radians/rpy_degrees/q_wxyz">`. By default `@rotation_type=rpy_radians` so it would be backward compatible.

FirefoxMetzger, 2021-08-01:
Thanks for pointing that out @azeey . I skimmed the PR and it does look like the better choice than `<value degrees="True/False">`.

EricCousineau-TRI, 2022-01-10:
Closing given merge of #689