gazebosim/sdformat #655: SDFormat10 package not available on Ubuntu

varunagrawal, 2021-08-04:
## Environment
* OS Version: 20.04
* Binary build for SDFormat10.

## Description
* Expected behavior: As per the README, running `apt-get install libsdformat10` should install the SDFormat package from Ubuntu's repositories.
* Actual behavior: `apt-get` cannot find the package.

## Steps to reproduce

1. Take a fresh version of Ubuntu 20.04 (I used a docker image to verify this).
2. Follow the README and run `sudo apt-get -y install libsdformat10`.
3. See the failure.

## Output

```sh
root@447c4bd99237:/# apt -y install sdformat10
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package sdformat10
```

varunagrawal, 2021-08-04:
My guess is that we need to run these extra commands before trying to install:

```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys D2486D2DD83DB69272AFE98867170598AF249743
sudo apt-get update

sudo apt-get -y install libsdformat10  # This should work now
```

If so, this should be added to the README.

azeey, 2021-08-06:
Yes, we currently only have `libsdformat6` on Ubuntu 20.04, so if other versions are needed, it's necessary to add the `packages.osrfoundation.org` repo. Do you mind creating a PR with your suggestion?

PR #679 (2021-08-30): Update Ubuntu install docs

# 🦟 Bug fix

Fixes #655

## Summary

Added instructions to add the correct apt repository to get sdformat binaries to install on Ubuntu.

## Checklist
- [X] Signed all commits for DCO
- [X] Added tests
- [X] Updated documentation (as needed)
- [X] Updated migration guide (as needed)
- [X] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [X] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [X] While waiting for a review on your PR, please help revi [... cut]
closed 2021-08-30 by azeey