gazebosim/sdformat #704: DOC: possibly inaccurate description on `//model/include`

FirefoxMetzger, 2021-09-20:
Currently, the spec for `//model/include` states that `Included resources can only contain one 'model', 'light' or 'actor' element.`:

https://github.com/ignitionrobotics/sdformat/blob/5182c0e679ea19da0dfd356e9821b2d146e23d2c/sdf/1.8/model.sdf#L47

However, `//model` can neither have `//light` nor `//actor` as children. I think that the description here is talking about the `//world` element, for which the above is valid, and that the description here should drop `light` and `actor` (as only nested models are okay within a `//model`).

If this is the case, I can submit a PR.

azeey, 2021-09-21:
Yes, I think that's correct. `//model/include` should not include `light` or `actor`.

scpeters, 2021-09-27:
should we allow including a light under a `link`?

FirefoxMetzger, 2021-09-27:
@scpeters You have a better overview of how people use SDFormat. Would there be an actual need for this?

If I want to include a light I can always write a small nested model ala

```xml
<model name="my_model">
  <link name="my_link">
    <light name="my_light"/>
  </link>
</model>
```

Which I can place as desired using `//include/@placement_frame` combined with a fixed joint in the parent. This is more verbose than a direct include inside `//link` but doesn't require extra parser logic to support.

On the opposite side, why only include `//light` and not any of the other children of `//link` like `//visual`, `//collision`, etc.? (Disclaimer: I never quite understood why `//light` can be included. The element itself doesn't seem very complex so I don't get why I'd want to outsource it.

closed 2021-09-27 by chapulina

scpeters, 2021-09-28:
> On the opposite side, why only include `//light` and not any of the other children of `//link` like `//visual`, `//collision`, etc.? (Disclaimer: I never quite understood why `//light` can be included. The element itself doesn't seem very complex so I don't get why I'd want to outsource it.

The reason we can include a `//light` is so they can be added to a world file. For example, the default world file for `gazebo` ([empty.world](https://github.com/osrf/gazebo/blob/gazebo11/worlds/empty.world#L6)) includes the `sun` light source.

I didn't think through my earlier comment. As you point out, the `//link/light` is treated similarly to `//visual`, `//collision`, etc. in not being `//include`able, so it's currently consistent.