gazebosim/sdformat #705: Clarify the behavior of `//model/model/@static`

FirefoxMetzger, 2021-09-20:
What happens if a model contains a _nested_ model that sets `//model/static` to `true`, i.e., how does `//model/model/static` behave if true? Will the child's `__model__` frame be attached to the parent model's `__model__` frame, or will it be attached to the `world` frame as per the docs and the spec?

The latter would mean that any static model will be left behind as the parent model moves, which strikes me as odd/unexpected behavior. However, I do realize that the former might cause trouble for some physics engines since they may use special optimizations if they know that a model will be static and that they may break if the object does end up moving.

If someone could clarify the behavior here, I can update the description of `//model/static` to reflect the discussion here.

azeey, 2021-09-24:
The nested/child model will be static and it's `__model__` frame will be attached to the `world`. What happens to the parent model depends the contents of the parent model. If the parent model has one if its child links as a canonical link, then the parent `__model__` frame will be attached to that. eg.
```xml
<world name="W">
  <model name="PM">
    <link name="L1"/>
    <model name="CM">
      <static>true</static>
    </model>
  </model>
</world>
```

Otherwise, it will use the child model's `__model__` frame as the canonical link, which means it will be indirectly attached to `world`.

```xml
<world name="W">
  <model name="PM">
    <model name="CM">
      <static>true</static>
    </model>
  </model>
</world>
```

>  Will the child's __model__ frame be attached to the parent model's __model__ frame

The child's `__model__` frame cannot be attached to the parent's `__model__` whether it's static or not. The reverse can happen as I described above.

FirefoxMetzger, 2021-09-25:
> Otherwise, it will use the child model's __model__ frame as the canonical link

Oh? This can happen? I think this would be definitely useful to add, since I was under the impression that `@canonical_link` may only refer to links. As a consequence, I thought that a parent model without any links would choose the first nested child's `@canonical_link` as its canonical link, or - alternatively - the first link inside a nested child. However, I don't have any good motivation for this behavior. The documentation that I am aware off so far doesn't explicitly say what should happen here.

It is slightly off-topic, but I am actually wondering the same thing about `//links` that have no chain of `//joint`s that connect them to the world. This is related to the `//must_be_base_link` question that @diegoferigo asked a few days ago. I will leave a comment there.

> The child's __model__ frame cannot be attached to the parent's __model__ whether it's static or not.

I get it now I think, thanks for explaining. I was aware that `Parent::__model__` is not in the scope of `Parent::Child::__model__` and hence `Parent::Child::__model__` can't use it as a `@relative_to` target or attach itself to it. Technically, the same is true for `world`, since it is outside any model's scope and `static` allows to make an exception for it; hence, me wondering if this behavior interacts with nesting somehow.

In sum, I'd update the description to

```
If set to true, the model is immovable; i.e., a dynamics engine will not update its position. This will also overwrite this model's `@canonical_link` and instead attach the model's implicit frame to the world's implicit frame. This holds even if this model is nested (or included) by another model instead of being a direct child of `//world`.
```

I'd also make an addition to `@canonical_link` of the form 

```
If unset or set to an empty string, the first `//link` listed as a direct child of this model is chosen as the canonical link. If the model has no direct `/link` children, its will instead be attached to the first nested (or included) model's implicit frame.
```

azeey, 2022-01-24:
Closed by #713

closed 2022-01-24 by azeey