gazebosim/sdformat #718: `/test/sdfmodel_nested_static_model.sdf` doesn't define a `@canonical_link`

FirefoxMetzger, 2021-09-28:
This could be related to me not having full knowledge of the `@canonical_link` resolution rules.

This is the SDF in question:

```python
```

FirefoxMetzger, 2021-09-28:
I just noticed that the same happens for `/test/sdf/root_multiple_models.sdf`, so I assume that there is another rule to `@canonical_link` that I am unaware of. In this case, the SDF looks like this

```xml
<?xml version="1.0" ?>
<sdf version="1.7">
  <model name="robot1">
    <link name="link"/>
  </model>
  <model name="robot2">
    <link name="link"/>
  </model>
  <model name="last_robot">
    <static>true</static>
  </model>
</sdf>
```

Perhaps setting `//static` to `true` inside a model means that the `@canonical_link` will be set to that model's `__model__` frame?

scpeters, 2021-09-28:
Our [documentation on sdformat.org](http://sdformat.org/tutorials?tut=pose_frame_semantics&cat=specification&#model-frame-and-canonical-link) discusses the different rules for canonical links in static and non-static models. We didn't want models to require a link, but we needed something to which the implicit model frame (`__model__`) is attached. For non-static models, at least one link is required so it can have a canonical link and well-defined `__model__` frame. For static models, the `__model__` frame can be attached to the `world` frame, so a canonical link is not strictly needed.

I believe this was originally added to the proposal for SDFormat 1.7 in https://github.com/ignitionrobotics/sdf_tutorials/commit/6f34b021a1fd8efade453e95f58a9dd502e326f3, but there isn't much explanation there.

FirefoxMetzger, 2021-09-28:
Thanks for explaining, @scpeters . I was/am aware of the attachment rules in the presence of `//static`, but I do seem to have misunderstood that last part of the documentation (`The name of the canonical link can be specified in the //model/@canonical_link attribute. If it is not specified, then the first //link tag in the model is chosen as the canonical link.`) as `@canonical_link` will always have a value.

Taking the perspective of "`@canonical_link` is what the `__model__` frame is attached to" it would essentially mean that the `@canonical_link` of a `//static` model is `world`, right? As far as I can see, this would be consistent with the current behavior and might make the logic a bit cleaner: 

1. If `//static` -> `@canonical_link` is `world`
2. (el)If `@canonical_link` is set -> `@canonical_link` is the frame set in the attribute
3. (el)if direct `//link` in `//model` -> `@canonical_link` is the frame of the first direct `//link`
4. (el)if nested model/include -> `@canonical_link` is `__model__` of the first nested child.
5. Pose of model can't be determined (raise error/exception)

Perhaps this would become even more obvious if `@canonical_link` is renamed to `@attached_to` (which would imply frames, not just `//links`) combined with some documentation detailing how the default is chosen if it is not set explicitly.

FirefoxMetzger, 2021-09-28:
an alternative name could be `@canonical_frame` ... mainly to remove the association with `//link` and to create an association with `//frame`.