gazebosim/sdformat #719: `test/sdf/joint_axis_xyz_normalization.sdf` appears to be missing `link6`

FirefoxMetzger, 2021-09-28:
A joint in the SDF `joint_axis_xyz_normalization` mentions a `link6`:

https://github.com/ignitionrobotics/sdformat/blob/91d0d0192c0658cc883d20cd24fd317885bbf6e9/test/sdf/joint_axis_xyz_normalization.sdf#L34-L40

However the link isn't defined in that file:

https://github.com/ignitionrobotics/sdformat/blob/91d0d0192c0658cc883d20cd24fd317885bbf6e9/test/sdf/joint_axis_xyz_normalization.sdf#L4-L9

Is it missing and should be added, or is this intentional?

FirefoxMetzger, 2021-09-28:
The same applies to `joint_axis_infinite_limits.sdf`:

https://github.com/ignitionrobotics/sdformat/blob/91d0d0192c0658cc883d20cd24fd317885bbf6e9/test/sdf/joint_axis_infinite_limits.sdf#L43-L55

scpeters, 2021-10-04:
> The same applies to `joint_axis_infinite_limits.sdf`:
> 
> https://github.com/ignitionrobotics/sdformat/blob/91d0d0192c0658cc883d20cd24fd317885bbf6e9/test/sdf/joint_axis_infinite_limits.sdf#L43-L55

This is clearly an error but is especially interesting because this file is loaded into an `sdf::Root` object without errors in [test/integration/joint_axis_dom.cc](https://github.com/ignitionrobotics/sdformat/blob/sdformat10_10.6.0/test/integration/joint_axis_dom.cc#L225-L234). I've confirmed that it fails `ign sdf --check` (https://github.com/scpeters/sdformat/commit/b47f9bf650525551c25e1f2729bda0d3008e0526), so there are some checks that we don't perform when loading an `sdf::Root` object.  Currently the value of `//joint/child` is validated in `FrameSemantics.cc` since it is needed to build the graphs, but `//joint/parent` is not directly used in any of the graphs, so it is not currently checked.

So, in addition to fixing this specific `.sdf` file, should we add stricter checks on the `//joint/parent` values when loading an `sdf::Root`?

* If so, where should the checks occur?
    * `Joint::Load`?
    * `FrameSemantics.cc`?

cc @azeey

scpeters, 2021-10-04:
> So, in addition to fixing this specific `.sdf` file, should we add stricter checks on the `//joint/parent` values when loading an `sdf::Root`?
> 
> * If so, where should the checks occur?
>   
>   * `Joint::Load`?
>   * `FrameSemantics.cc`?

Actually, I think it couldn't be done in `Joint::Load`, since the other objects may not have been loaded yet. So maybe `Model::Load`?

FirefoxMetzger, 2021-10-04:
I actually don't have a lot of insight into the C++ bindings of SDFormat, so you're ideas here are likely better than mine.

On a high level, however, `ign sdf --check` should have a C++ backend that the ruby script calls to perform the actual validation. Anything wrong with calling that immediately after the code enters `sdf::Root`? This way there would be only one code path for validation (sounds like there is currently some duplication). It would also make the binding code cleaner because it can focus on creating the objects and can assume that the SDF itself is valid.

It would also open the door into a `validate=false` flag that skips validation in case it impacts runtime...

scpeters, 2021-10-04:
> On a high level, however, `ign sdf --check` should have a C++ backend that the ruby script calls to perform the actual validation. Anything wrong with calling that immediately after the code enters [...]

this is the c++ code for `ign sdf --check`: https://github.com/ignitionrobotics/sdformat/blob/sdf11/src/ign.cc#L34-L99

the first thing it does is load an `sdf::Root` object, but then it calls many other helper functions. we could consider adding these to `sdf::Root::Load` I suppose

scpeters, 2021-10-05:
I've targeted a fix for `sdf10` in #726

scpeters, 2021-10-05:
fix for `sdf11` in #727

closed 2021-10-15 by scpeters