gazebosim/sdformat #742: Invalid boolean value

MatthijsBurgh, 2021-11-09:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build: Binary 9.7.0
 
<details>
  
```
Error [Param.cc:329] Invalid boolean value
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error [Param.cc:329] Invalid boolean value
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error [Param.cc:329] Invalid boolean value
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Dbg [parser.cc:574] Converting a deprecated source[data-string].
Dbg [Converter.cc:68] Version[1.5] to Version[1.7]
  Please use the gz sdf tool to update your SDF files.
    $ gz sdf -c [sdf_file]
Dbg [parser.cc:574] Converting a deprecated source[data-string].
Dbg [Converter.cc:68] Version[1.5] to Version[1.7]
  Please use the gz sdf tool to update your SDF files.
    $ gz sdf -c [sdf_file]
```

</details>

## Description
* Expected behavior: Not thrown any warning and load the nested models correctly
* Actual behavior: Error on invalid boolean value in the block120 model. It results in the blocks not being loaded. Heightmap and thin wall model are loaded.

I load gazebo with a world file, `impuls`. I have tracked down that the errors are related to the `static` tag of the `block120` model, which is included in `impuls/walls`. These are still SDF 1.6 models. I have also tried with SDF 1.7 models. Which produce the same errors. Including the `parse as version 1.7 failed` error.

The error only occurs when the `static` is set to `true`. When set to `false` no errors are shown and models are loaded in gazebo. Though I want the models to be static.

I think the bug might have been introduced in https://github.com/ignitionrobotics/sdformat/commit/42f8fad0d3f9a13ae8d17fb13692acdc80fff966. As older version of sdformat9 did not give me any issues. Bug could also already been in 9.6.0, but 9.5.0 and older was 100% not giving me any issues.

## Steps to reproduce
Opening gabezo with a world_model file.

World model file: https://github.com/tue-robotics/ed_object_models/blob/0a157094f2f79996b7a26659216689e43f1ca216/models/impuls/model.sdf
Walls file: https://github.com/tue-robotics/ed_object_models/blob/0a157094f2f79996b7a26659216689e43f1ca216/models/impuls/walls/model.sdf
Block file: https://github.com/tue-robotics/ed_object_models/blob/0a157094f2f79996b7a26659216689e43f1ca216/models/robotics_testlabs/block120/model.sdf
(Thin wall file: https://github.com/tue-robotics/ed_object_models/tree/0a157094f2f79996b7a26659216689e43f1ca216/models/impuls/thin_wall)

## Output
```
Error [Param.cc:329] Invalid boolean value
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error [Param.cc:329] Invalid boolean value
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error [Param.cc:329] Invalid boolean value
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error Code 8 Msg: Error reading element <static>
Error Code 8 Msg: Error reading element <model>
Error Code 8 Msg: Unable to parse sdf element[sdf]
Error Code 8 Msg: Error reading element <static>
Error Code 8 Msg: Error reading element <model>
Error Code 8 Msg: Unable to parse sdf element[sdf]
Error Code 8 Msg: Error reading element <static>
Error Code 8 Msg: Error reading element <model>
Error Code 8 Msg: Unable to parse sdf element[sdf]
```

azeey, 2021-11-10:
@jennuine Do you mind taking a look?

jennuine, 2021-11-12:
Hi @MatthijsBurgh, I've looked into the issue and it turns out it is a bug in the process of flattening nested models (see https://github.com/ignitionrobotics/sdformat/issues/751).

> The error only occurs when the static is set to true or 1. When set to false no errors are shown and models are loaded in gazebo

In short, when the model is loaded into an `sdf::Element` the value of `<static>` becomes `0` or `1` for `false` or `true`. Since several of the included models (including the `block120`) have elements with `name='1'` the parser is mistakenly replacing the value of `<static>` to `[model_name]::1`, which is why the `Invalid boolean value` has occurred. When `<static>` set to false, it doesn't get replaced because it differs from the `name` attribute.

> I think the bug might have been introduced in 42f8fad. As older version of sdformat9 did not give me any issues. Bug could also already been in 9.6.0, but 9.5.0 and older was 100% not giving me any issues.

I am not sure how this was working before. I tested with `9.5.0` and a release done last year and the issue still occurs.

---

Unfortunately, finding a solution may take us some time. As a work around I suggest renaming element names to be something else other than a single numeric value (e.g., `0` or `1`). Also, I see several of the models have siblings with the same name (e.g., for `visual`s and `collision`s). To prevent any additional issues, we recommend updating the names of siblings to be unique since it's a requirement of the SDFormat specification.

I am going to close this issue since the bug is described in https://github.com/ignitionrobotics/sdformat/issues/751 but please feel free to reopen if you believe otherwise.

closed 2021-11-12 by jennuine

MatthijsBurgh, 2021-11-13:
I already started renaming the collisions and visuals in the SDF 1.7 versions. I will also change the names of the links.

I think it might be good to describe this bug somewhere else. So other people can find it more easily.

@jennuine Thanks for your efforts.